adafruit/circuitpython #4640: Update library .gitignore to be more robust

kattni, 2021-04-20:
We should update the `.gitignore` on the libraries to be more robust. There are plenty of good options to start with. We already include the PyCharm and VSCode directories in the `cookiecutter` version, but the current libraries do not all have that included. I think we should discuss here what to add before considering applying a patch to all the current libs to bring them up to date with our current working version. I will also organise and comment the final version so it's clearer.

There are an overwhelming number of options [here](https://github.com/github/gitignore) and we could start with one of these. [This site](https://www.toptal.com/developers/gitignore) allows you to build one that works for you, but it also produces ones that are massive.

**Please comment in this thread with any suggestion additions along with their purpose.**

Here is what we have right now in `cookiecutter`:
```
*.mpy
.idea
__pycache__
_build
*.pyc
.env
.python-version
build*/
bundles
*.DS_Store
.eggs
dist
**/*.egg-info
.vscode
```

dhalbert, 2021-04-20:
From my own global `.gitignore`:
```
# emacs temp file suffixes
*~
.#*
\#*#

# TAGS files (cross-reference tools)
GTAGS
GRTAGS
GPATH
TAGS

# Results for the checking of the Python coding style and static analysis
.mypy_cache
flake8_output.txt

# VIM files
*.swp
*.swo

# VS Code Settings
.vscode/       # Note /
```

jposada202020, 2021-04-20:
I will add `venv` that is used by python when crating virtual environments and

dhalbert, 2022-05-14:
Superceded by https://github.com/adafruit/cookiecutter-adafruit-circuitpython/pull/179

closed 2022-05-14 by dhalbert