adafruit/circuitpython #4644: Lockups with OV7670 camera at 24MHz

jepler, 2021-04-21:
This is a weird one, but both @jerryneedell and I have seen it.

If an OV7670-style camera is plugged into the Grand Central M4, and a 24MHz PWM signal is generated on pin D29, then "weird stuff™" happens. This causes a crash or hard-fault during camera initialization (before the new ParallelImageCapture object is even instantiated), and occasionally things that look like non-fatal memory corruption.  If the PWM signal is slower (nominal 16MHz, actually 17.mumble MHz) then things are OK.

The problem seems not to occur when the OV7670 is not connected.  The OV7670 starts generating input signals on PCC_CLK, PCC_DEN{1,2}, PCC_D{0-7} as soon as the external clock signal is applied.  No special functions should be enabled on those pins, though!

I have failed to track this down more closely.  As the desired clock rate for the OV7670 is 24MHz it would be good to discover and fix the problem.

daveythacher, 2021-04-22:
24MHz is enough for 30FPS of 640x480  using 565.  (510*784*30*2) You will probably deplete the entire RAM in this situation. This would require 600KB, which is more than the 256KB.

ladyada, 2021-04-22:
@daveythacher this is the PCLK rate, not related to FPS, we are not using video.

daveythacher, 2021-04-22:
> @daveythacher this is the PCLK rate, not related to FPS, we are not using video.

PCLK is correlated to data rate. CLK is correlated to internal clock. (Table 5) Some internal CLK is correlated to PCLK. (Table 5) A single frame will exceed RAM limit. The exact internal clock value needed is not clear me, however I did not read the entire thing. I failed to see internal osc, so I can only assume there is a correlation from CLK to PCLK. I saw the pull request which works off single frame capture. I had to guess this was what he was speaking of.

Point is you do not need 24MHz since you are not working with video and you are not working with whole frame. The min CLK per datasheet is 10MHz. (Table 4)

I guessed the 24MHz by assuming it was like VGA. Which Figure 6 suggests. Based on the description and Figure 6, I was able to  compute 24MHz. This however makes assumptions about register configurations. I do not know based on this issue what settings he used there. I could guess it was from the Adafruit_OV7670 code base, however suggest here appears to suggest new ImageCapture module.

Figure 6 suggests for a 640x480 frame you will have 510 Tlines which are 784 Tp. Tp is 1 or 2 depending on number of bytes for color scheme. If you have 30 frames you will have 1 / 24MHz for Tp min.

This mentions about three different issues. Most if not all actually have nothing to do with OV7670.
1) Issue were non-fatal memory corruption can occur with 24MHz.
2) 16MHz is not correct.
3) Pin mux behavior not as expected.
4) Is the input rate the problem. If I generate 24MHz by get back 2MHz will it continue to work?