adafruit/circuitpython #4657: Pi Pico (RP2040): `pulseio.PulseIn.deinit()` doesn't seem to release the pin hardware

eigenein, 2021-04-25:
### Firmware

```text
Adafruit CircuitPython 6.2.0 on 2021-04-05; Raspberry Pi Pico with rp2040
```

### Code 1

```python
from board import GP13
from pulseio import PulseIn

PulseIn(GP13).deinit()
PulseIn(GP13).deinit()  # Traceback 1
```

### Code 2

```python
from board import GP13
from pulseio import PulseIn

with PulseIn(GP13):
    pass
with PulseIn(GP13):  # Traceback 2
    pass
```

### Traceback 1

```python
Traceback (most recent call last):
  File "code.py", line 5, in <module>
ValueError: GP13 in use
```

### Traceback 2

```python
Traceback (most recent call last):
  File "code.py", line 6, in <module>
ValueError: GP13 in use
```

gamblor21, 2021-04-25:
Verified it myself. Took a look and the deinit() function needs to call `reset_pin_number(self->pin);`

(Looked at displayio.ParallelBus for a reference that also uses the PIO and allocates a pin)

gamblor21, 2021-04-25:
I was looking to quickly fix this but came across another issue that may be deeper. I added the `reset_pin_number` but now get the following:

```
p = pulseio.PulseIn(board.D11)
p = pulseio.PulseIn(board.D11)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: D11 in use
p.deinit()
p = pulseio.PulseIn(board.D11)
p = pulseio.PulseIn(board.D11)
>>> 
```
It seems however the PulseIn PIO is being allocated after the first `reset_all_pins` the condition used to check if a pin is in use is never set again. This gets in the RP2040 SDK. I don't have a debugger set up so it isn't too easy for me to check out what isn't being set at the moment.

On a side note PulseIn is still called the underlying state machine construct and not the common_hal one.

closed 2021-04-25 by gamblor21

eigenein, 2021-04-25:
That was really quick, thank you! :)