adafruit/circuitpython #4658: Cannot use Control-C to interrupt code.py after sending string

jerryneedell, 2021-04-25:
I am using the code below on a neopixel_trinkey_m0 to send a string when either of the trinkey cap touch pads are touched.

I have found that it works fine, but if I have the REPL open then use then trigger it to send more than once, then I am no longer able to interrupt the execution of code.py from the REPL with a Control-C. Also if I try to exit my "screen" session on Linux, I am unable to close the session.  The code keeps running fine.. If I move my focus to a different window, I can see the string being printed on each touch.  I have to press RESET to get out of code,py.

I'm not sure if this is an issue with the library or with usb_hid.

I also had this same issue with a PIRKEY_M0 using irremote to trigger sending the string.

```
from adafruit_hid.keyboard import Keyboard
from adafruit_hid.keycode import Keycode
from adafruit_hid.keyboard_layout_us import KeyboardLayoutUS
import neopixel
import board
import time
import usb_hid
import touchio 

TOKEN = "0123456789012345678901234567890123456789"

led = neopixel.NeoPixel(board.NEOPIXEL,4,brightness=.3)

# The keyboard object!
time.sleep(1)  # Sleep for a bit to avoid a race condition on some systems
keyboard = Keyboard(usb_hid.devices)
keyboard_layout = KeyboardLayoutUS(keyboard)  # We're in the US :)

pad1 = touchio.TouchIn(board.TOUCH1)
pad2 = touchio.TouchIn(board.TOUCH2)
print("Ready! - touch to send code ")

while True:
    led.fill(0x100000)    # set LETs dimm  RED while waiting
    if pad1.value or pad2.value:
        led.fill(0x001010) # Change LED color-- cyan is pretty
        keyboard_layout.write(TOKEN)
        time.sleep(1)

```

to invoke the issue;
load this code as code.py
open terminal session to REPL
touch the cap touch pad to send string  - leds turn blue while sending (it helps to move your focus to another window to see the string)
repeat - touch again
in REPL session - try to interrupt with Control-C - it will not respond

jerryneedell, 2021-04-25:
More information:
If I shorten the string to 22 characters or fewer, the problem does not occur.
At 23 or above, it does.
at 23 characters it would sometimes stop responding to control-c after the string was sen 3 times instead of 2 as with the longer string.  On a few occasions it dd respond to a control-c but after a delay of a few seconds.

jerryneedell, 2021-04-25:
update:
with a string of 10 characters, the response to the control-C appears to be delayed but it does responsd even after several (more than 5) string transmissions.
There is not a clear "break point" ... yet

dhalbert, 2021-08-17:
I retested the shorter test program on a Metro M4 and Trinket M0 on the tip of `main`, and it seems to work OK. There have been some recent fixes that I think might have fixed this. Please reopen if you can reproduce. Thanks.

closed 2021-08-17 by dhalbert