adafruit/circuitpython #4659: [RP2040] `pulseio.PulseIn` isn't able to catch long pulses

eigenein, 2021-04-25:
## Firmware

```text
Adafruit CircuitPython 6.2.0 on 2021-04-05; Raspberry Pi Pico with rp2040
```

## `code.py`

**Note:** `GP16` and `GP17` are directly connected with a wire.

```python
from board import GP16, GP17
from digitalio import DigitalInOut, Direction
from pulseio import PulseIn
from time import sleep

trigger = DigitalInOut(GP16)
trigger.direction = Direction.OUTPUT
trigger.value = False

echo = PulseIn(GP17, maxlen=10, idle_state=False)

def trigger_pulse(seconds: float):
    print(f"Triggering ~{seconds:.6}s pulse…")
    trigger.value = True
    sleep(seconds)
    trigger.value = False
    print("Done:", [echo[i] for i in range(len(echo))])

sleep(1.0)  # giving myself enough time to connect to the serial
trigger_pulse(0.000500)
trigger_pulse(0.001000)
trigger_pulse(0.002000)
trigger_pulse(0.003000)
trigger_pulse(0.004000)
trigger_pulse(0.005000)
trigger_pulse(0.006000)
trigger_pulse(0.007000)
trigger_pulse(0.008000)
trigger_pulse(0.009000)
```

## Output

```text
code.py output:
Triggering ~0.0005s pulse…
Done: [663]
Triggering ~0.001s pulse…
Done: [663, 224]
Triggering ~0.002s pulse…
Done: [663, 224, 1150]
Triggering ~0.003s pulse…
Done: [663, 224, 1150, 2211]
Triggering ~0.004s pulse…
Done: [663, 224, 1150, 2211, 3965]
Triggering ~0.005s pulse…
Done: [663, 224, 1150, 2211, 3965]
Triggering ~0.006s pulse…
Done: [663, 224, 1150, 2211, 3965]
Triggering ~0.007s pulse…
Done: [663, 224, 1150, 2211, 3965]
Triggering ~0.008s pulse…
Done: [663, 224, 1150, 2211, 3965]
Triggering ~0.009s pulse…
Done: [663, 224, 1150, 2211, 3965]

Code done running.
```

## Consequences

[Adafruit HC-SR04 library](https://github.com/adafruit/Adafruit_CircuitPython_HCSR04) doesn't work well on Pi Pico since it uses `pulseio` to measure echo pulse durations. You're not able to measure any distance longer than ~0.5m.

SuperThunder, 2021-05-31:
I think the merged PR 4668 fixes issues I had with Feather RP2040 not picking up the 9ms and 4.5ms pulse of NEC8 infrared remote :)

In 6.2 those pulses didn't appear at all, now in 7.0.0-alpha.2-705-gc80e25307 I am picking up the pulses and able to semi-succesfully funnel them into the decoding function I lifted from [micropython_ir](https://github.com/peterhinch/micropython_ir/blob/master/ir_rx/nec.py) and get the right command code out.

I was also unable to use adafruit_irremote with 6.2 and Feather RP2040, so I assume that probably works now too or is close to working (I am not sure how to update the library file from 6.2 to 7.0 to test)

tannewt, 2021-06-01:
Which library are you trying to move to 7.0? If it is an mpy then you'll need to snag one from a 7.x bundle or release zip.

tannewt, 2021-06-01:
Fixed by #4668. (comment if not and I'll reopen.)

closed 2021-06-01 by tannewt

SuperThunder, 2021-06-04:
> Which library are you trying to move to 7.0? If it is an mpy then you'll need to snag one from a 7.x bundle or release zip.

I was trying out adafruit_irremote and assumed there weren't any builds for 7.0 because they don't appear on the [libraries page](https://circuitpython.org/libraries), but I found the [CircuitPython Bundle releases](https://github.com/adafruit/Adafruit_CircuitPython_Bundle/releases) page after your comment.  
It does work now and recognizes the command from my remote about 80% of the time.