adafruit/circuitpython #4669: Neotrellis - i2s "no pull up found on SDA or SCL"

hamslabs, 2021-04-26:
**Firmware**

```python
Adafruit CircuitPython 6.2.0 on 2021-04-05; Adafruit QT Py M0 with samd21e18
and
Adafruit CircuitPython 6.2.0 on 2021-04-05; Adafruit Feather RP2040 with rp2040
```

**Code/REPL**

```python
import time
 
from board import SCL, SDA
import busio
from adafruit_neotrellis.neotrellis import NeoTrellis
from adafruit_neotrellis.multitrellis import MultiTrellis
 
# create the i2c object for the trellis
i2c_bus = busio.I2C(SCL, SDA)
```

**Behavior**

```python
Traceback (most recent call last):
  File "code.py", line 11, in <module>
RuntimeError: No pull up found on SDA or SCL; check your wiring
```

**Description**

- I wired the i2c port from to the fingers on the Neotrellis board since I didn't have a cable with the right connector for that connection. I checked it with a meter to verify that it is connected properly and there are no shorts.
 
**Additional Info**

- I looked at SDA and SCL with a logic analyzer and they are both low the entire time the code runs.

dhalbert, 2021-04-26:
This sounds like there is no power being supplied to the NeoTrellis, because the the pullups should be pulling those lines up. TOr there may be something wrong on the NeoTrellis board. This is a support issue. Could you please start a thread in https://forums.adafruit.com, and include pictures of your wiring? Thanks. You can post a comment here later with a link to the forum thread.

closed 2021-04-26 by dhalbert

hamslabs, 2021-04-26:
I posted this in the forums yesterday but got no replies. The subject line is "i2c problem with Feather RP2040 and Neotrellis". I'll post pics of the wiring there right now.