adafruit/circuitpython #4670: [RP2040] [Pi Pico] [macOS] CIRCUITPY volume broken in `bd5a3a3`

eigenein, 2021-04-26:
### `INFO_UF2.txt`

```text
UF2 Bootloader v2.0
Model: Raspberry Pi RP2
Board-ID: RPI-RP2
```

### OS

macOS Big Sur 11.2.3 (20D91)

### Description

Latest build that works is `adafruit-circuitpython-raspberry_pi_pico-en_GB-20210411-c333acd.uf2`, I've done the bisect actually flashing the images onto the device.

With these builds the volume doesn't show up:

- `adafruit-circuitpython-raspberry_pi_pico-en_GB-20210412-bd5a3a3.uf2` – this is the build following the latest working one. Commit reference: bd5a3a3.
- `adafruit-circuitpython-raspberry_pi_pico-en_GB-20210426-7cf8f79.uf2` – this is the latest build at the moment.

With the broken builds I can still see `/dev/cu.usbmodem14401`, but `screen /dev/cu.usbmodem14401` gives me nothing but an empty unresponsive terminal.

dhalbert, 2021-04-26:
@tyomitch tagging you for interest: bd5a3a3 is from #4583.

tyomitch, 2021-04-26:
> @tyomitch tagging you for interest: bd5a3a3 is from #4583.

Intriguing! I'll try to get a Pi Pico to debug on.

@eigenein do the broken builds still boot into main.py, i.e. is it only the USB interface that no longer works, or does the whole thing freeze up during boot?

eigenein, 2021-04-26:
@tyomitch Hmm, it's just got way weirder.

I've uploaded a simple blink+print script, uploaded the bd5a3a3 firmware and the volume has appeared just fine. Same with 2a78a51 and today's 7cf8f79. With the latest firmware I was able to re-attach the device multiple times to USB and always see the mounted volume.

That wasn't the case while I was doing the bisect between `6.2.0` and the latest build; actually I've re-uploaded bd5a3a3 and c333acd multiple times to ensure I've found the correct commit.

So frankly I'm lost now, maybe there's no issue at all and we can simply close that.

eigenein, 2021-04-28:
[Last few builds](https://adafruit-circuit-python.s3.amazonaws.com/index.html?prefix=bin/raspberry_pi_pico/en_GB/) are also working fine.

@tyomitch Shall I close the issue for now, unless you want me to test something on the device?

tyomitch, 2021-04-28:
> @tyomitch Shall I close the issue for now, unless you want me to test something on the device?

Please feel free to close.

eigenein, 2021-04-28:
Sure. I'll re-open it should it pop up again.

closed 2021-04-28 by eigenein