adafruit/circuitpython #4671: Native Tensorflow Lite support

hierophect, 2021-04-26:
[Tensorflow Light for Microcontrollers](https://www.tensorflow.org/lite/microcontrollers) is a subset of Tensorflow Light designed specifically for use with microcontrollers, taking only 16 KB for the core runtime and requiring no system libraries or dynamically allocated memory. Adding these tools would allow for things like person detection, gesture sorting, voice recognition, and other tasks that use trained models.

Tensorflow Light is already implemented for Micropython on the [OpenMV](https://docs.openmv.io/library/omv.tf.html) and the [Sipeed MAix BiT](https://www.seeedstudio.com/Sipeed-MAix-BiT-for-RISC-V-AI-IoT-p-2872.html). It is available using platforms like Arduino and Mbed on a variety of other development boards.

This thread is for general discussion of this implementation and tracking potential issues with its implementation.

mocleiri, 2021-08-21:
I've built a micropython integration for tensorflow lite micro: https://github.com/mocleiri/tensorflow-micropython-examples

I've implemented all of the upstream examples except for magic wand.  Some are just running micropython code with hard coded sample data and others like micro_speech can run in real-time with audio sampled over i2s.

I'm progressively expanding what has been done and I'm interested in integrating with circuit python to expand the number of people that can use it to experiment with python on microcontrollers.

The main issue is firmware size.  The C++ api linker uses dead-code detection and only includes the associated code for the ops you are using which drastically reduces the size of the final firmware.

In a scripting language this is not possible because when we make the firmware its impossible to know which ops are needed,  

At the moment the overhead for libtensorflow-microlite.a is 1.5 MB impact on the firmware size.  This is fine for ESP32 because most have 4 MB flash but for many boards that is way to much. 

I think it should be possible to add in compile time flags to control what subset of ops are included by default to shrink down the firmware size.

What is a common board with 2MB or more of flash that I can start with trying to build for?

hierophect, 2021-08-21:
@mocleiri my first instinct would be the STM32H7 (probably H743), which we have basic support for, is used by OpenMV, and has a precedent for being used for this kind of application. However, a number of Circuitpython boards use external flash, including the ESP32S2, the i.MX RT, and the RP2040, all of which could theoretically support the firmware size you're mentioning. It'd probably be nice to pair it with a platform that supports a camera or audio features.

mocleiri, 2021-08-22:
Thanks for your feedback.

I'm tracking circuitpython integration on: https://github.com/mocleiri/tensorflow-micropython-examples/issues/27

I'm looking at reducing the ops and firmware size in:
https://github.com/mocleiri/tensorflow-micropython-examples/issues/28

I'll start with the Cortex M7 boards like openmv and teensy 4.1 and then into the others you listed.

I'll ping back on this issue once there are firmwares to test.

tannewt, 2021-08-23:
Thanks for looking into this @mocleiri.

mocleiri, 2021-09-02:
I found out that the circuitpython stm32 port doesn't support C++ which is a problem but I think it can be worked around.

Would it be possible to intake this micropython commit to add c++ support to stm32?

https://github.com/micropython/micropython/commit/97960dc7deb7a0e691fca5944402cd03386b744b

The other option is to build the c++ parts externally and link them into the firmware.  I'm already doing this for the tensorflow bits but it could be expanded to also include the micropython c++ binding layer.

It took me a while due to linker issues (due to differences of esp32 being based on freertos versus the stm32 port being bare metal) but I was able to build the microlite firmware today for the [Nucleo-H743ZI2/](https://os.mbed.com/platforms/ST-Nucleo-H743ZI2/) in micropython as a user_c_module.

I was able to flash and connect to repl and it looks like the microlite module is present.  I will test and verify its working correctly by running the hello_world and person detection examples.

I built the libtensorflow-microlite.a in my project via a [ci job](https://github.com/mocleiri/tensorflow-micropython-examples/blob/feature/31-support-stm32-boards/ci/build_tensorflow_cortex_m_generic.sh), [(action)](https://github.com/mocleiri/tensorflow-micropython-examples/actions/runs/1180820822) and then copied the file locally and used it to build the firmware.

Now that it works I will add a ci job in my project for the micropython build for this board and then come back to the circuitpython build.  I will create a pull request with the circuitpython changes and we can iterate on how exactly to structure the c++ parts so that circuitpython can be built and be used.

One issue is that on the tensorflows side it builds a different version of the library for cortex m0, m4 and m7 including with the different floating point numbers and cmsis_nn options.

For esp32 it was easy 1:1 tflite to micropython but for stm32 I think some kind of scheme where the board can specify the kind of libtensorflow-microlite.a it should link to will be needed. I would improve the naming of the artifacts published and then the circuitpython ci scripts can be updated to use a github action like:https://github.com/dawidd6/action-download-artifact to download the correct kind of library from my repository.

tannewt, 2021-09-02:
I'd be happy to see C++ support integrated into our build files. I've not used C++ for embedded though so I'm not sure what exactly would need to change.

Why are you planning on linking to the prebuilt .a files? I'd prefer to build it directly into the binary if we can. That way we can tweak the compilation options and maybe use LTO.

mocleiri, 2021-09-03:
Using the static library was a shortcut which worked on esp32 so I didn't have to know how it all worked on the tensorflow side.  

In getting to the answer on how to link it I built most of such an approach to build everything from the micropython side.

I will continue and incorporate into my eventual pull request.

mocleiri, 2021-11-18:
Thanks for the useful article and breakpoint advice.

I found out my issue with stm32 is most likely related to it being bare metal and me not doing the right thing in terms of wiring up the standard library entry points (malloc/free, c++ new/delete) to use the micropython heap instead.

I was able to get the Adafruit feather RP2040 to build just fine:  https://github.com/mocleiri/tensorflow-micropython-examples/actions/workflows/build_rp2.yml

I'm part way through getting tflm to build as part of my microlite module.  The core library builds but I need to do more work still to re-integrate the audio frontend which wraps the tflm micro_speech c spectrogram generating code.

I'm going to try a mimxrt port since I have a teensy 4.1 (although it also looks to be bare metal like stm32) and then after that circle back to this issue and taking what ever I have working through to integrate with Circuit Python

tannewt, 2021-11-18:
Cool. Thanks for the update! Sorry I can't help with the C++ stuff.

CaptainHaddock2016, 2024-05-07:
Any updates on this? This would be undoubtedly a significant feature to add... plus simple tensorflow lite models can run on most Adafruit boards.

tannewt, 2024-05-13:
> Any updates on this? This would be undoubtedly a significant feature to add... plus simple tensorflow lite models can run on most Adafruit boards.

Not from me. What are you looking to do with tensorflow lite specifically? I haven't seen any use that is really useful.

nstielau, 2025-03-20:
Gesture detection on XIAO nRF52840 (Sense)!  That is a pretty useful usecase IMO ;)

AndreasTheCat, 2025-09-29:
@nstielau any chances for collaborating on liteRT on cp? I'm in ESP32 & predictive maintenance.