adafruit/circuitpython #4682: AttributeError: 'PixelBuf' object has no attribute '_transmit'

makermelissa, 2021-04-28:
When resuming from light sleep, the code crashes with this error:
```
Traceback (most recent call last):
  File "code.py", line 44, in <module>
  File "/lib/adafruit_funhouse/__init__.py", line 122, in enter_light_sleep
AttributeError: 'PixelBuf' object has no attribute '_transmit'
```

I'm not sure what happened since I had previously tested this, but will fix it as soon as I can figure out what is going on.

makermelissa, 2021-04-28:
According to @kattni, this issue has been reportedly been seen on the Saola and QT Py M0 Haxpress. I observed in on the FunHouse.

anecdata, 2021-04-28:
This appeared for me after 7.0.0 (Saola, but suspect it isn't port-dependent), never saw it in 6.2.0 on any device.

Neradoc, 2021-04-28:
`fill` does not cause the bug in the same version I encountered it:
```py
Adafruit CircuitPython 7.0.0-alpha.1-844-g9e183bbbb on 2021-04-27; Adafruit QT Py M0 Haxpress with samd21e18
>>> import board
>>> import neopixel
>>> pix = neopixel.NeoPixel(board.NEOPIXEL,1)
>>> pix.fill((0,255,0))
```

hierophect, 2021-04-28:
I'm running into this now that all the 6.2 mpy files are deprecated for 7.0.0

jerryneedell, 2021-04-29:
FYI -- I was able to "workaround" this but using adafruit_pypixelbuf on adafruit_dotstar instead of _pixelbuf
That is, in adafruit_dotstart, just import adafruit_pypixelbuf as _pixelbuf
It now works with 7.0.0 (tip of main)

dunkmann00, 2021-04-29:
This must have something to do with merging the newer versions of Micropython (#4646, #4649). '_transmit' exists in the Neopixel and Dotstar subclasses, not in _Pixelbuf, which is the superclass. So for some reason instead of searching the subclass type for the transmit function and then searching the _Pixelbuf type, it would seem Circuitpython is only searching _Pixelbuf. 

This would also support why 'fill' would work, as it is a function from _Pixelbuf. It is interesting using adafruit_pypixelbuf works, since that would indicate the problem has to do with subclassing a native type from Circuitpython, but not something written in Python.

If @tannewt has plans to continue to merge the newer versions of Micropython perhaps this will be fixed in one of those?

makermelissa, 2021-04-29:
Makes sense. Last I heard he is planning to continue. If it doesn't fix it, then it would make more sense to address this issue after that is finished so it doesn't break again.

kattni, 2021-04-29:
Yes, the plan is to merge though the current version. As there is no 7.0.0 release yet, I agree we should wait to address this issue until we're done merging MicroPython.

tannewt, 2021-04-29:
@jepler Is looking at this. `_transmit` is usually on the subclass (not PixelBuf) so it's likely a bug with the attribute lookup.

jerryneedell, 2021-05-02:
I believe this was resolved by #4688 -- OK to close?

closed 2021-05-02 by jepler