adafruit/circuitpython #4686: Doing almost anything at the repl hard-faults.

jepler, 2021-04-29:
I pulled out a Metro M0 to do some testing with, and the current main branch just crashes whenever I do "almost anything" at the repl.

```
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.

Press any key to enter the REPL. Use CTRL-D to reload.

Adafruit CircuitPython 7.0.0-alpha.1-929-g52589691b on 2021-04-29; Adafruit Metro M0 Express with samd21g18
>>> 
>>> import sys

[tio 10:57:51] Disconnected
[tio 10:57:53] Connected
Not running saved code.

You are in safe mode: something unanticipated happened.
CircuitPython core code crashed hard. Whoops!
Crash into the HardFault_Handler.
                                 Press any key to enter the REPL. Use CTRL-D to reload.

Adafruit CircuitPython 7.0.0-alpha.1-929-g52589691b on 2021-04-29; Adafruit Metro M0 Express with samd21g18
```

When I debug a local build the only ,
```
Breakpoint 1, HardFault_Handler () at supervisor/port.c:578
578	    reset_into_safe_mode(HARD_CRASH);
(gdb) where
#0  HardFault_Handler () at supervisor/port.c:578
#1  <signal handler called>
#2  0x200005c8 in _pystack ()
#3  0x00015242 in mp_execute_bytecode (code_state=0x200005ac <_pystack>, inject_exc=<optimized out>) at ../../py/vm.c:177
#4  0x20002022 in ?? ()
```

If my code is
```
>>> 1+1
```
I get a different traceback, but still a hardfault:
```
Breakpoint 1, HardFault_Handler () at supervisor/port.c:578
578	    reset_into_safe_mode(HARD_CRASH);
(gdb) where
#0  HardFault_Handler () at supervisor/port.c:578
#1  <signal handler called>
#2  0x0000013a in ?? ()
#3  0x00015242 in mp_execute_bytecode (code_state=0x200005ac <_pystack>, inject_exc=<optimized out>) at ../../py/vm.c:177
#4  0x0000d494 in fun_bc_call (self_in=0x20001fb0, n_args=0, n_kw=0, args=<optimized out>) at ../../py/objfun.c:285
#5  0x0000a69e in mp_call_function_n_kw (fun_in=0x20001fb0, n_args=0, n_kw=0, args=0x0) at ../../py/runtime.c:651
#6  0x0000a664 in mp_call_function_0 (fun=<optimized out>) at ../../py/runtime.c:625
#7  0x0002e4d0 in parse_compile_execute (source=<optimized out>, input_kind=<optimized out>, exec_flags=22, result=0x0) at ../../lib/utils/pyexec.c:114
#8  0x0001a454 in pyexec_friendly_repl () at ../../lib/utils/pyexec.c:566
#9  run_repl () at ../../main.c:532
#10 main () at ../../main.c:599
```
note that the line in mp_execute_bytecode that the crash is attributed to is the same line of mp_execute_bytecode both times:
```
(gdb) info locals
nlr = {prev = 0x20007f5c, ret_val = 0x0, regs = {0x15243 <mp_execute_bytecode+46>, 0x200005ac <_pystack>, 0x20001fb0, 0x20000350 <mp_state_ctx>, 0x200005ac <_pystack>, 0xbbfeffef, 0xf5dffdef, 0xffbfcbbf, 
    0xfbfefebb, 0x20007e68}, pystack = 0x200005cc <_pystack+32>}
ip = <optimized out>
ulab = <optimized out>
fastn = <optimized out>
exc_stack = <optimized out>
exc_sp = 0x200005bc <_pystack+16>
outer_dispatch_loop = <optimized out>
dispatch_loop = <optimized out>
entry_table = {[manually omitted]}
```

jepler, 2021-04-29:
A grand central m4 doesn't seem to be affected.  6.2.0 works on the Metro M0 Express, at least to the extent I tested it.

jerryneedell, 2021-04-29:
On QTPU M0 Haxpress, I tried 
```
help('modules')
```
at the REPL -- it went do SAFE Mode due to " Crash into HArdFault_Handler"

ajs256, 2021-04-29:
Cannot repro on a RasPi Pico running 
```
Adafruit CircuitPython 7.0.0-alpha.1-929-g52589691b on 2021-04-29; Raspberry Pi Pico with rp2040
```
I tried to do `help("modules")`, `1+1`, and `import sys`. All worked as expected. No safe mode or HardFault. This is probably a SAMD-specific problem.

jerryneedell, 2021-04-29:
Yes, I've had no issues on rp2040 or esp32s2 boards.

commit 41b273e0 (2021-04-30): py/vm.c: Restore lost bits of MICROPY_OPT_COMPUTED_GOTO_SAVE_SPACE

This fixes a problem where boards that enabled this (most SAM D21s)
would crash on the first bytecode instruction.

Closes: #4686
closed 2021-04-30 by tannewt