adafruit/circuitpython #4697: mpy-cross build issue on 64bit Raspberry Pi OS

Idirianuk, 2021-05-01:
Hopefully, this issue is of relevance here. My apologies if this is an issue which should not have been raised here.

Hi,

At the moment, on my setup, at least, it is not possible to build mpy-cross.

It has been building up until, and including 43840363c2ac1709ff0ed456f042a7b31c82db41
After, and including 76033d5115889d779debfddc14c6990df31de799, building fails with the following error:

`make: Entering directory '/home/john/code/circuitpython/mpy-cross'
`
`
Use make V=1, make V=2 or set BUILD_VERBOSE similarly in your environment to increase build verbosity.
`
`
QSTR updated
`
`
../py/nativeglue.c:242:5: error: ‘nlr_push’ undeclared here (not in a function); did you mean ‘nlr_jump’?
`
` 
    nlr_push,
`
`
     ^~~~~~~~
`
`
     nlr_jump
`
`
make: *** [../py/mkrules.mk:55: build/py/nativeglue.o] Error 1
`
`
make: Leaving directory '/home/john/code/circuitpython/mpy-cross'
`

There is a reference to this at:

[https://github.com/micropython/micropython/issues/4666](url)

My current experience prevents me from shedding further light or fixing (even for myself) this issue, hence this issue being raised.

Thank you all for your hard work and most excellent output. :)

ladyada, 2021-05-01:
try
https://github.com/micropython/micropython/issues/4666#issuecomment-479748439

Idirianuk, 2021-05-01:
Thanks for your reply. :)

I took a look at that last night. I was unsure as to where to apply the fixes mentioned and remain so.

As I said, if I checkout at the earlier merge, all is well. All is not well after the merge of MP 1.11, so something has changed which breaks things now.

Interestingly, I am able to build mpy-cross from the downstream micropython.

Hopefully, that didn't come across as a bit sharp, as that's certainly not my intention. :)

p.s Love my three Feather 2040s! :)

ladyada, 2021-05-01:
kk good to know!

jepler, 2021-05-02:
It looks like when we merge in MicroPython 1.12 (#4693) this should be fixed.  We should add a static-aarch64 build of mpy-cross when this is done.

Idirianuk, 2021-05-02:
Albeit temporary, I've ovecome this by cheating most heinously.

Purists, look away... I built mpy-cross in my clone of the micropython repo then copyied the executable into my clone of circuitpython. :P

Idirianuk, 2021-05-05:
Following merge of 1.12, this seems to be fixed now.

closed 2021-05-05 by Idirianuk

jepler, 2021-05-05:
In #4712 we added building mpy-cross for aarch64 so this regression should be noticed in the future.  Thanks again1