adafruit/circuitpython #4699: Implement _dht_read() mini-module to do DHT-sensor-specific reads

dhalbert, 2021-05-01:
As mentioned in https://github.com/adafruit/circuitpython/pull/4689#issuecomment-830649832 and to support #4696, it would be convenient to have a small C module that can do the DHT-style read protocol, without needing the entirety of `PulseIn`

It would implement just the basic read, like [this code](https://github.com/adafruit/DHT-sensor-library/blob/f339feb6b6865f4f818bf1ab9cf091f2172cd5c6/DHT.cpp#L231-L388) in [the DHT sensor library](https://github.com/adafruit/DHT-sensor-library/).

deshipu, 2021-05-01:
Why not call it ``_1wire_read``, as the protocol is called?

ladyada, 2021-05-01:
hm cause it isnt 1 wire (?) and weve got something with that name already

dhalbert, 2021-05-02:
DHT is _a_ one-wire protocol but is not the "official" "Dallas" one-wire protocol, so it's not compatible. DHT returns 5 bytes; Dallas returns 8 bytes, and I don't know whether the timing is similar in any way.

deshipu, 2021-05-02:
Sorry, I was misinformed.