adafruit/circuitpython #4700: Safe mode crash when initializing UART with tx=PA31 or PA11 on SAMD21

deshipu, 2021-05-01:
Adafruit CircuitPython 7.0.0-alpha.1-815-ged374a89f on 2021-04-24; Fluff M0 with samd21e18

The following code:
```python
import busio
import board

uart = busio.UART(tx=board.RX)
```

results in:
```
Auto-reload is off.
Running in safe mode! Not running saved code.

You are in safe mode: something unanticipated happened.
CircuitPython core code crashed hard. Whoops!
Crash into the HardFault_Handler.
                                 Press any key to enter the REPL. Use CTRL-D to reload.
```

dhalbert, 2021-05-01:
Does it work on 6.2.0? Is there anything connected on those pins right now?

deshipu, 2021-05-01:
Same behavior whether anything is connected or not.
I will try with 6.2.0.

deshipu, 2021-05-01:
6.2.0 doesn't seem to work on the Fluff M0 at all. On 6.1.0 I get:

```
code.py output:
Traceback (most recent call last):
  File "code.py", line 4, in <module>
ValueError: Invalid pins
```

which seems to be correct.

emwdx, 2021-06-21:
I am having the same problem on a Metro M4 express with CircuitPython 6.3.0.

```
import board
import busio

uart = busio.UART(board.D2, board.D3, baudrate=115200)
```

In the REPL, I also get the Invalid pins ValueError on the uart line.

PR #5026 (2021-07-21): Fix crash when UART construct fails
After the script stops with the exception thrown the final gc_sweep
will call any finalizers and they usually call deinit. deinit on
invalid objects can wreak havoc by changing random memory or
(hopefully) crashing. This fixes ensures the object is deinited
until initialization succeeds and the object is valid.

Do the same fix for I2C and SPI too.

Fixes #4700 and fixes #5005
commit fece0fb4 (2021-07-21): Fix crash when UART construct fails

After the script stops with the exception thrown the final gc_sweep
will call any finalizers and they usually call deinit. deinit on
invalid objects can wreak havoc by changing random memory or
(hopefully) crashing. This fixes ensures the object is deinited
until initialization succeeds and the object is valid.

Do the same fix for I2C and SPI too.

Fixes #4700 and fixes #5005
closed 2021-07-21 by microDev1