adafruit/circuitpython #4705: Use libgcc.a compiled with -Os instead of the default -O2 on boards that are tight on space

dhalbert, 2021-05-03:
The official ARM gcc toolchain compiles `libgcc.a` as `-O2`. The `arch` Linux ARM toolchain uses `-Os`. We discovered that building on `arch` saved a kB or two (at least).

We can consider replacing the stock `libgcc.a` in the ARM toolchain with a `-Os` version to make room on the smallest boards for new features.

tannewt, 2022-03-08:
@dhalbert You look into this a bit right? What did you run into?

dhalbert, 2022-03-08:
I tried to build `libgcc.a` as part of building the ARM gcc toolchain from their supplied sources, so I could then change `libgcc`s build flags and make sure I got a library that would be cleanly substitutable. Despite following the directions carefully, I was unable to get the toolchain to build far enough to build that library. I think I tried this twice. Perhaps it will be easier with the 11.2 sources

tannewt, 2022-03-08:
I know that they changed the way they build their archive. I think they use Linode CI now. (I saw the info on the embedded.fm slack.)

commit e7d72b1e (2022-08-05): Use libgcc from Arch on SAMD21 (it's smaller.)

Fixes #4705
closed 2022-08-05 by tannewt