adafruit/circuitpython #4710: Writing to UART from Pi Pico is a noop

markrawls, 2021-05-05:
**Firmware**

```python
Adafruit CircuitPython 6.2.0 on 2021-04-05; Raspberry Pi Pico with rp2040
```

**Code/REPL**

```python
import board
import busio

from time import sleep

uart = busio.UART(tx=board.GP0, rx=board.GP1, baudrate=115200)

while True:
    print(uart.write('Hello, world\n'.encode()))  # 0, every time
    sleep(1)
```

**Behavior**

Unfortunately, nothing. The UART interface reports 0 bytes written, the computer attempting to receive the messages prints out an empty byte string. When manually resetting the Pico I can get a null byte sent down the line, but that's it.

**Description**

Attempting to send bytes via UART from the Pico to a Pi 4 (or other system) does not actually write anything. The return value of `uart.write()` also indicates no bytes written. This appears to be unaffected by baud rate, as I have tried many different values with no success.

microDev1, 2021-05-05:
@markrawls This issue was fixed recently. Try the latest build from [S3](https://adafruit-circuit-python.s3.amazonaws.com/index.html?prefix=bin/raspberry_pi_pico/en_US/).

markrawls, 2021-05-05:
Running with boot message `Adafruit CircuitPython 7.0.0-alpha.1-1004-ga2ae50350 on 2021-05-04; Raspberry Pi Pico with rp2040`, the return value from `uart.write()` is correct, but nothing is actually getting transmitted down the wire.

microDev1, 2021-05-05:
I am unable to replicate this issue. Can you try the following code. 
Connect `GP0` & `GP1` together. The expected output is `hello` in `REPL` terminal.
```python
import time
import board
import busio
import digitalio

led = digitalio.DigitalInOut(board.LED)
led.direction = digitalio.Direction.OUTPUT

s = bytearray("hello\n\r".encode())
uart = busio.UART(board.GP0, board.GP1, baudrate=115200)

while True:
    uart.write(s)	
    data = uart.read(10)  # read up to a byte
    # print(data)  # this is a bytearray type

    if data is not None:
        led.value = True
 
        # convert bytearray to string
        data_string = ''.join([chr(b) for b in data])
        print(data_string, end="")
 
        led.value = False

```
If you get the expected output than there is something wrong at the receiver's end.

markrawls, 2021-05-05:
I connected GP0 and GP1 together and used that code (modified to use `board.GP28` instead of `board.LED` as that's where I have a status LED hooked up) and saw no output. I change the wire to the one that was successfully receiving data before, but still not output whatsoever. Uncommenting that print statement showed `None` as the result of every read.

dhalbert, 2021-05-05:
Try GP4 and GP5 with the 7.0.0 version; maybe you have a bad pin?

markrawls, 2021-05-05:
GP4 and GP5 did work with the 7.0 version, looks like my GP1 pin is dead. That's unfortunate, but I'm glad it was just a simple flaw along those lines.

closed 2021-05-05 by markrawls

dhalbert, 2021-05-05:
Check whether your USB-serial converter (if that's what you're using) is set for 5V rather than 3.3V levels.

markrawls, 2021-05-05:
It's connected directly to a Raspberry Pi 4B on GPIO pins 14 & 15, which are 3.3V.