adafruit/circuitpython #4714: CircuitPython 6.2.0 does not boot on Espruino Pico

valholl, 2021-05-06:
I've tried to flash adafruit-circuitpython-espruino_pico-en_US-6.2.0.bin on the Espruino Pico following the advanced reflashing instructions on https://www.espruino.com/Pico#advanced-reflashing , but after what seemed a succesful flash the board didn't react in any way (nothing was logged in dmesg).

I have also tried 7.0.0-alpha.1 with the same result

On the same board I have successfully flashed micropython, following the same instructions.

I've also found this post on the forum https://forums.adafruit.com/viewtopic.php?f=60&t=174183&p=849038&hilit=espruino#p849038 so I'm opening this issue.

Please let me know if you need me to do any further test; I'm using linux on the host I'm using to flash the board.

dhalbert, 2021-05-06:
Could you try some earlier versions: 6.1.0, 6.0.0, 5.3.1? That will help us narrow it down. You can download them here:
https://adafruit-circuit-python.s3.amazonaws.com/index.html?prefix=bin/espruino_pico/

dhalbert, 2021-05-06:
@hierophect This is an STM32F401xE. Do you have any guess of why this would have stopped working?

I don't have one to try.

hierophect, 2021-05-06:
Could be something got messed up with the oscillators in the course of the v6 updates. I'll check it out.

hierophect, 2021-05-06:
@valholl Hi, want to give this binary from my fix in #4719 a test? This fixes the problem on my Pico. Normally I'd point to the build artifacts but CI is being weird and didn't generate any.
[firmware.bin.zip](https://github.com/adafruit/circuitpython/files/6437789/firmware.bin.zip)

valholl, 2021-05-07:
@hierophect it works also on my pico, thanks!

PR #4719 (2021-05-07): Espruino Pico: Remove settings for DNP oscillator
The LSE on the Espruino Pico is actually DNP, which isn't documented on the schematic. This was causing a clock failure after the 6.0 STM32 timekeeping reworks.

Resolves #4714
closed 2021-05-07 by dhalbert