adafruit/circuitpython #4723: f string implementation

kevinjwalters, 2021-05-07:
The f string impementation from https://github.com/adafruit/circuitpython/pull/2690 does not appear to implement setting parameters with `{}` syntax. An example from `6.2.0` on a Feather nRF52840 Express:

```
>>> val = 1.2345
>>> digits = 2
>>> f"{val:.{digits}f}"
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
KeyError: digits
```

I noticed this whilst trying to help out with [Adafruit Forums: How to display decimals on Bitcoin price?](https://forums.adafruit.com/viewtopic.php?f=60&t=178978).

tannewt, 2021-05-07:
Internally f-strings are implemented by translating to a `.format` call. That makes this tough to do I believe.

kevinjwalters, 2021-05-07:
This works:

```
Adafruit CircuitPython 6.2.0 on 2021-04-05; Adafruit Feather nRF52840 Express with nRF52840
>>>
>>> val = 1.2345
>>> digits = 2
>>> "{val:.{digits}f}".format(val=val, digits=digits)
'1.23'
```

kevinjwalters, 2021-05-19:
Other fstring tickets: #4780

Other format tickets: #4776

Neradoc, 2022-05-21:
This was fixed at some point after 7.0:
```py
Adafruit CircuitPython 7.2.0 on 2022-02-24; Adafruit CLUE nRF52840 Express with nRF52840
>>> 
paste mode; Ctrl-C to cancel, Ctrl-D to finish
=== val = 1.2345
=== digits = 2
=== f"{val:.{digits}f}"
=== 
'1.23'
```

dhalbert, 2022-05-21:
Thanks, we can close.

closed 2022-05-21 by dhalbert