adafruit/circuitpython #4746: changing USB descriptors on ESP32S2 crashes

Neradoc, 2021-05-11:
Tested on the Feather S2. I expected that trying to use more endpoints than available results in reset to safe mode. Here are a few `boot.py` that cause issues. (Manual safe mode recovers).
```py
import usb_cdc
usb_cdc.enable(console=True, data=True)
```
- reset loop (status LED alternates purple and black)

```py
import usb_hid
usb_hid.disable()
import usb_cdc
usb_cdc.enable(console=True, data=True)
```
- code.py runs
- device connected seen on USB (ioreg) but no drive, no serial

### Those work as expected:
```py
import usb_hid
usb_hid.disable()
```
```py
import usb_cdc
usb_cdc.enable(console=True, data=False)
```
```py
import usb_hid
usb_hid.disable()
import storage
storage.disable_usb_drive()
import usb_cdc
usb_cdc.enable(console=True, data=True)
```

closed 2021-05-14 by tannewt

Neradoc, 2021-05-15:
It's not fixed, I have both error cases above do a reset loop on latest on ESP32-S2.

dhalbert, 2021-05-15:
I tested on the ESP32-S2 with your examples and did not get a reset loop, hmm.

Neradoc, 2021-05-15:
Oh it works when I erase_flash and install with the Circuitpython bin, but not if I install the UF2 bootloader first.

dhalbert, 2021-05-15:
I only tested with the .bin. I just re-tested and it goes into safe mode with `USB devices need more endpoints than are available.` as I expected with a boot.py:
```py
import usb_cdc
usb_cdc.enable(console=True, data=True)
```

If boot.py is empty and you use the UF2 version does it work?

Could you compare the initial prompt with the version number between the .bin. and the .uf2 version to make sure they are the same build? Thanks.

There isn't anything in my USB impl that should care whether it's UF2 or .bin.

Neradoc, 2021-05-15:
I tried with the artifacts, with the the latest S3, I fixed my toolchain to be able to build locally, same result every time. I tried the bin from the same local build, goes to safe mode. Empty boot.py, no boot.py, or the boot.py listed above to work do work as expected.
TinyUF2 uses USB, could it leave it a state where there's some endpoints cleanup needed ?

dhalbert, 2021-05-15:
See #4766 for problem diagnosis.

todbot, 2021-07-05:
I'm still having this problem.  On an ESP32S2 MagTag 2.9", I do the following:
```
esptool.py  --port  /dev/tty.usbmodem01 erase_flash
esptool.py --port  /dev/tty.usbmodem01 write_flash 0x0 ~/adafruit-circuitpython-adafruit_magtag_2.9_grayscale-en_US-20210703-cece649.bin
```
and I get a fresh CP7 install. I then create the following `boot.py`:

```py
# boot.py
import usb_midi
usb_midi.disable() 
print("disable MIDI")
import usb_hid
usb_hid.disable()
print("disable HID")
```

Reset the board and the above works. I see those print statements in `boot_out.txt`.  However, if I change `boot.py` to this:

```py
# boot.py
import usb_midi
usb_midi.disable() 
print("disable MIDI")
import usb_hid
usb_hid.disable()
print("disable HID")
import usb_cdc
usb_cdc.enable(console=True, data=True)
print("enable CDC console & data")
```
Upon reset, the board goes into a boot loop and no USB device appears in the USB devices list.  Upon first reset, the D4 LED of the MagTag flashes yellow four times (and then maybe briefly blue?).  The only recourse is to repeat the `erase_flash` and `write_flash` steps with `esptool.py`.

Neradoc, 2021-07-06:
I was able to reproduce with the Magtag (`7.0.0-alpha.3-495-gcece6492a` built locally, without tinyuf2).
- The Magtag does not appear on USB.
- It is able to be recovered by pressing boot on the yellow blink on reset. (Manually go to safe mode).

The Feather S2 in the same conditions reboots into safe mode (with the second `boot.py`).
```
You are in safe mode because:
USB devices need more endpoints than are available.
```

tannewt, 2021-07-08:
@todbot please open a new issue. Closed issues are hard to track and I suspect this might be a different problem from the original. (My guess is that the safe mode restart isn't working.)

PR #4754 (2021-05-14): Fix endpoint checking; Clean up safe mode printing
Fixes #4746.
Fixes #4752 (STM32 HID support).

- Endpoint count checking was off by one.
- `CIRCUITPY_USB_{HID,MIDI,MSC}_ENABLED_DEFAULT` default values were not set conditionally.
- Clean up `print_safe_mode_message()` to simplify logic, reduce size, and divide up safe-mode reasons more accurately.
- Redo logic of printing startup messages in `run_code_py()` to not print the safe-mode message twice, clean up newlines, and be more consistent about whitespace.
- Shortened and rephrased some messages slightly.
EDIT:
- Count in and out endpoints  [... cut]