adafruit/circuitpython #4752: STM32(F4) USB HID support

Gigahawk, 2021-05-12:
I've bought a Feather STM32F405 Express and was expecting I could get it to act as a USB keyboard, but turns out USB_HID (and quite a lot of other features) aren't implemented for STM32 chips yet.

Is there any effort being made to implement USB_HID support?

dhalbert, 2021-05-12:
USB HID (and MIDI) are theoretically available, but they are not turned on. The problem is that the USB physical interface (PHY) that is connected to the USB connector only has 4 endpoint pairs. One pair is reserved for control purposes, one pair is used for mass storage (CIRCUITPY), and two pairs (well, 1.5), are used for the CDC serial (REPL connection). So that leaves none for anything else.

There is another USB PHY on the chip, a HS-capable one with more endpoints, but it is not connected to anything. It is not connected on the PyBoard either. I'm not sure why, but I think there was some reason.

With dynamic USB descriptors available in CircuitPython, it would be possible to disable the REPL and enable HID. It wouldn't be that pleasant a development experience. You might consider a different board with more endpoints, such as an RP2040 one.

Gigahawk, 2021-05-12:
Is there documentation on how to disable the REPL?

Or better yet, route it to the serial port so I can just hook up a separate adapter to get to it.

dhalbert, 2021-05-12:
This is all really, really new, and we're still getting the edges smoothed off, but we will think about enabling the possibility of HID while at the same time leaving it off at runtime by default on the STM32F4 boards. See the `usb_hid` and `usb_cdc` doc here: https://circuitpython.readthedocs.io/en/latest/shared-bindings/index.html.

tannewt, 2021-05-13:
> There is another USB PHY on the chip, a HS-capable one with more endpoints, but it is not connected to anything. It is not connected on the PyBoard either. I'm not sure why, but I think there was some reason.

Most of the STM chips have HS peripherals but no HS PHY. They require a separate UTMI chip for the HS PHY. That's why they aren't used very often.

Gigahawk, 2021-05-14:
@dhalbert thanks, calling `usb_cdc.disable()` in boot.py makes `usb_hid` work.

The cherry on top would be access to the REPL via the dedicated serial pins, is there any way to do that?

PR #4754 (2021-05-14): Fix endpoint checking; Clean up safe mode printing
Fixes #4746.
Fixes #4752 (STM32 HID support).

- Endpoint count checking was off by one.
- `CIRCUITPY_USB_{HID,MIDI,MSC}_ENABLED_DEFAULT` default values were not set conditionally.
- Clean up `print_safe_mode_message()` to simplify logic, reduce size, and divide up safe-mode reasons more accurately.
- Redo logic of printing startup messages in `run_code_py()` to not print the safe-mode message twice, clean up newlines, and be more consistent about whitespace.
- Shortened and rephrased some messages slightly.
EDIT:
- Count in and out endpoints  [... cut]
closed 2021-05-14 by tannewt