adafruit/circuitpython #4753: ulab array corruption with slices

kevinjwalters, 2021-05-12:
I noticed this on CircuitPython 6.2.0 on a CLUE (nRF52840):

```
Adafruit CircuitPython 6.2.0 on 2021-04-05; Adafruit CLUE nRF52840 Express with nRF52840
>>> import ulab
>>> import ulab.numerical
>>> import ulab.filter
>>> data = ulab.ones(10000)
>>> fir_taps = ulab.array([0.15, 0.2, 0.3, 0.2, 0.15])
>>> filtered_data = ulab.filter.convolve(data, fir_taps)[6:-6]
>>> ulab.numerical.sum(filtered_data)
9992.0
>>> len(filtered_data)
9992
>>> a = []  ### append to an array to write and use up free memory
>>> for idx in range(1_000_000):
...     a.append("AYMABTU")
...
...
...
Traceback (most recent call last):
  File "<stdin>", line 2, in <module>
MemoryError: memory allocation failed, allocating 131072 bytes
>>> del a
>>> ulab.numerical.sum(filtered_data)
9882.01
>>> filtered_data[0:10]
array([2.98023e-08, 0.0, 1.19209e-07, 0.0, -6.61744e-24, 0.0, -2.64698e-23, 0.0, 1.70141e+38, 0.0], dtype=float32)
```

I'd guess it's related to premature GC of the `ndarray` returned by `ulab.filter.convolve()`. That would be inappropriate as the _slice_ is a view onto that data as I understand it.

kevinjwalters, 2021-05-13:
@v923z may be interested in this.

v923z, 2021-05-13:
@kevinjwalters 

> That would be inappropriate as the slice is a view onto that data as I understand it.

This is correct, but perhaps the first question is, whether this has anything to do with `ulab`, and `convolve`. I believe, especially, if this is a problem with the garbage collection, we should strip this to the most trivial example that produces such behaviour. Something like this: 

```python
import math

_data = [1.0] * 10000
data = _data[6:-6]
sum(data)
a = []
for idx in range(1_000_000):
    a.append("AYMABTU")

del a
sum(data)
```
If this still works, I would then do 

```python

_data = np.array([1.0] * 10000)
data = _data[6:-6]
np.sum(data)
a = []
for idx in range(1_000_000):
    a.append("AYMABTU")

del a
np.sum(data)
```
Do you get similar results, if you try to initialise `a = ['AYMABTU'] * 1_000_000`?

Also, when you have this
```python
ulab.numerical.sum(filtered_data)  ### no longer 9992.0 !!
9882.01
```
can you find out, where the data are corrupted? At the beginning, or the end of the array?

kevinjwalters, 2021-05-13:
First part:

```
Adafruit CircuitPython 6.2.0 on 2021-04-05; Adafruit CLUE nRF52840 Express with nRF52840
>>>
>>> import math
>>> _data = [1.0] * 10000
>>> data = _data[6:-6]
>>> sum(data)
9988.0
>>> a = []
>>> for idx in range(1_000_000):
...     a.append("AYMABTU")
...
...
...
Traceback (most recent call last):
  File "<stdin>", line 2, in <module>
MemoryError: memory allocation failed, allocating 131072 bytes
>>> del a
>>> sum(data)
9988.0
```

Although perhaps this is closer to my example:

```
>>> import math
>>> _data = [1.0] * 10000
>>> data = _data[6:-6]
>>> del _data
>>> sum(data)
9988.0
>>> a = []
>>> for idx in range(1_000_000):
...     a.append("AYMABTU")
...
...
...
Traceback (most recent call last):
  File "<stdin>", line 2, in <module>
MemoryError: memory allocation failed, allocating 131072 bytes
>>> del a
>>> sum(data)
9988.0
```

I added a `del _data` here too:

```
Adafruit CircuitPython 6.2.0 on 2021-04-05; Adafruit CLUE nRF52840 Express with nRF52840
>>> import ulab as np
>>> _data = np.array([1.0] * 10000)
>>> data = _data[6:-6]
>>> del _data
>>> np.sum(data)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'module' object has no attribute 'sum'
>>> np.numerical.sum(data)
9988.0
>>> a = []
>>> for idx in range(1_000_000):
...     a.append("AYMABTU")
...
...
...
Traceback (most recent call last):
  File "<stdin>", line 2, in <module>
MemoryError: memory allocation failed, allocating 131072 bytes
>>> del a
>>> np.numerical.sum(data)
9988.0
```

v923z, 2021-05-13:
@kevinjwalters Thanks for bringing up the issue; it has been fixed in https://github.com/v923z/micropython-ulab/pull/388, and the fix backported to the `legacy` branch: https://github.com/v923z/micropython-ulab/releases/tag/1.7.8.

@jepler Thanks for digging to the root of the problem!

dhalbert, 2022-05-14:
Fixed as reported above.

closed 2022-05-14 by dhalbert