adafruit/circuitpython #4760: Dict order of items is incorrect

makermelissa, 2021-05-14:
Tested using CircuitPython 6.2.0 on the FunHouse.

```
test = {"a": "A", "b": "B", "c": "C"}
>>> test
{'c': 'C', 'a': 'A', 'b': 'B'}
```

Here it is in CPython:
```
>>> test = {"a": "A", "b": "B", "c": "C"}
>>> test
{'a': 'A', 'b': 'B', 'c': 'C'}
```

The CPython result is what I would expect for both.

Neradoc, 2021-05-14:
Is it a bug or an enhancement ?
Dict order was not guaranteed until python 3.6 and Micropython was based on 3.4 I believe.
You would do something like that (if you use the dict notation you'll lose the order).
```py
from collection import OrderedDict
OrderedDict([("a","A"),("b","B"),("c","C")])
```

dhalbert, 2021-05-14:
It's not in MicroPython (yet or ever): https://github.com/micropython/micropython/issues/6170

makermelissa, 2021-05-14:
Oh, ok. I guess this is an enhancement then.

tannewt, 2021-05-14:
There is collections.OrderedDict but its look ups are linear and therefore slower.

makermelissa, 2021-11-03:
We can close this if it's not in the plans to add. I think the `collections.OrderedDict` solution is workable.

tannewt, 2021-11-03:
I'm fine leaving it open. It is something we'd want at some point to match CPython.