adafruit/circuitpython #4761: REPL/console via hardware UART

dhalbert, 2021-05-14:
Provide support for accessing the REPL over UART. There are some compile switches for this, but I don't know how complete it is on the inside. There was support long ago because of the ESP8266. The BLE-only workflow would use a similar internal path.

One request: https://github.com/adafruit/circuitpython/issues/4752#issuecomment-841111515

Related to #721.

tannewt, 2021-05-14:
Yup, I may rethink this with the BLE work. It may involve giving UARTs a lifetime similar to displays.

fkpwolf, 2022-01-14:
I also want this feature since virtula/emulated UART is not stable when device usb is plugin and out. I can't debug device if error happened at the time.

tannewt, 2022-01-14:
@fkpwolf Please file a separate issue for what is unstable. It shouldn't be.

tannewt, 2022-03-17:
We do have this at build time through DEBUG_UART. We don't currently allow boot.py to set it up though.