adafruit/circuitpython #4766: ESP32-S2: cannot go into safe mode when using TinyUF2

dhalbert, 2021-05-15:
7.0.0 latest, with TinyUF2 in use only.

ESP32-S2 does not go into safe mode when requested to do so on a reset.

```python
>>> import microcontroller
>>> microcontroller.on_next_reset(microcontroller.RunMode.SAFE_MODE)
>>> microcontroller.reset()
```
On reconnect, CircuitPython does not report it's in safe mode.
```
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.

Press any key to enter the REPL. Use CTRL-D to reload.

Adafruit CircuitPython 7.0.0-alpha.2-570-g689ec8622 on 2021-05-14; Adafruit Metro ESP32S2 with ESP32S2
```

We also see this when it tries to force a safe mode reset, for instance when `boot.py` has specified devices that cause it to run out of endpoints. [Normally that is reported](https://github.com/adafruit/circuitpython/issues/4746#issuecomment-841584527) by a specific safe-mode reset.

This inability to go into safe mode occurs only when TinyUSB is in use. If the .bin version of CircuitPython is used, it goes into safe mode as expected:
```
Auto-reload is off.
Running in safe mode! Not running saved code.

You are in safe mode because:
The `microcontroller` module was used to boot into safe mode. Press reset to exit safe mode.

Press any key to enter the REPL. Use CTRL-D to reload.

Adafruit CircuitPython 7.0.0-alpha.2-570-g689ec8622 on 2021-05-14; Adafruit Metro ESP32S2 with ESP32S2
```

Tagging @hathach and @microDev1 for interest.
Other possibly related issues, which are different: #3988, #4304, #3884.

ThomasAtBBTF, 2021-05-19:
I really would appreciate if you would also look at the:
`microcontroller.on_next_reset(microcontroller.RunMode.BOOTLOADER)`

microDev1, 2021-07-19:
I just tested this with TinyUF2 0.5.0 and this seems to be fixed.

dhalbert, 2021-07-19:
This should be fixed now with the latest TinyUF2. Update your TinyUF2 bootloader to at least 0.5.0, following the instructions here: https://learn.adafruit.com/adafruit-magtag/install-uf2-bootloader. Thanks @microDev1 for testing and @hathach for the TinyUF2 fix.

closed 2021-07-19 by dhalbert