adafruit/circuitpython #4771: atmel-samd DEBUG UART support

EmergReanimator, 2021-05-17:
**Firmware**

atmel-samd based boards.

```python
```

**Code/REPL**

Not applicable

```python
```

**Behavior**

Not applicable

```python
```

**Description**

- There are many board which does have a USB port to run REPL terminal.
-  It would be great to provide DEBUG UART interface instead for such boards.

**Additional Info**

EmergReanimator, 2021-05-17:
I made the changes in existing code to support DEBUG UART on atmel-samd based boards.

```
CircuitPython 7.0.0-alpha.2-586-g0ac107b73-dirty on 2021-05-17; ATSAME51J20A custom board
```

I would like to submit my changes, but I would to clarify some doubts before doing this:

1. There is generic part which is relevant for all atmel-samX based boards.  But there is board specific part which currently resides inside of asf4_conf directory. The issue is that this configuration is really board specific, rather than chip specific. So I would like to introduce board specific configuration.
2. I have already allocated circuitpython fork for LPC17xx development, which is not yet complete. But I would like to deliver the changes in atmel-samd port now. May I create a branch from circuitpython and deliver atmel-samd changes there?

EmergReanimator, 2021-05-17:
suggested labels:

- atmel-samd
- enhancement

EmergReanimator, 2021-05-17:
Added patch for review, before delivering
[atmel-samd_uart.generic-part.patch.zip](https://github.com/adafruit/circuitpython/files/6496314/atmel-samd_uart.generic-part.patch.zip)

dhalbert, 2021-05-17:
Have you forked `adafruit/circuitpython` on GitHub? Do so and you can create multiple branches in your fork to handle all these changes. A GitHub pull request is the best way to submit code for review.

If this UART port only needs to exist when the CircuitPython VM is running, you should be able to implement it in a port and device-independent way. You can call the common-hal UART routines to set up the connection and then have `serial.c` and maybe other places direct output and take input from that UART connection. I don't think it needs to be atmel-samd specific.

dhalbert, 2021-05-17:
The `multiterminal` module did something like this in the past, but is not currently used. (I think I may have told you about this already.)

EmergReanimator, 2021-05-17:
> Have you forked `adafruit/circuitpython` on GitHub? Do so and you can create multiple branches in your fork to handle all these changes. A GitHub pull request is the best way to submit code for review.

Right! I did it this way. Created a branch atmel-samd in my fork.

EmergReanimator, 2021-05-21:
I believe this may be closed soon, since: #4772 passed the review.

commit 2b64b1d3 (2021-05-29): [resolves #4771] DEBUG UART supported on ATMSAME5x
commit b0dd0e34 (2021-05-29): Merge pull request #4772 from EmergReanimator/atmel-samd

[resolves #4771] DEBUG UART supported on ATMSAME5x
closed 2021-05-29 by jepler