adafruit/circuitpython #4776: No thousands separator in string format with float

anecdata, 2021-05-18:
**Firmware**

```python
Adafruit CircuitPython 6.2.0 on 2021-04-05; FeatherS2 PreRelease with ESP32S2
```

**Code/REPL & Behavior**

CPython allows combining comma and float string format specifiers:
```py
Python 2.7.17
>>> x = 1234567.55
>>> print("{:9,.1f}".format(x))
1,234,567.6
```
CircuitPython ignores the comma with a float (ignore the precision issue)
```py
Adafruit CircuitPython 6.2.0 on 2021-04-05; FeatherS2 PreRelease with ESP32S2
>>> x = 1234567.55
>>> print("{:9,.1f}".format(x))
1234567.4
```
CircuitPython handles comma fine with an `int`:
```py
Adafruit CircuitPython 6.2.0 on 2021-04-05; FeatherS2 PreRelease with ESP32S2
>>> x = 1234567
>>> print("{:,}".format(x))
1,234,567
```

richteel, 2022-04-13:
Just ran across this bug. I spent time checking and double checking if I missed something. Glad to finally run across this so I know I'm not losing my mind yet. ;-)

dhalbert, 2025-01-31:
Being fixed in https://github.com/micropython/micropython/pull/16480