adafruit/circuitpython #4777: Adafruit_CircuitPython_74HC165 Driver Library

woolseyj, 2021-05-18:
Adafruit (Kattni Rembor, Tony DiCola) wrote and provided the **Adafruit_CircuitPython_74HC595** driver library that can be used to add more digital outputs to a CircuitPython compatible board using the 74HC595 8-bit serial-in parallel-out (SIPO) shift register IC.

I have been working on a complementary project to add more digital inputs using the 74HC165 8-bit parallel-in serial-out (PISO) shift register IC. I copied and reversed/modified the original adafruit_74hc595.py library into a 74hc165 version that appears to be working well.

I realize that I could release this library into the community bundle, but I thought it would be more beneficial to the CircuitPython community if the Adafruit team took what I have, do whatever modifications they see fit, and released it in the main bundle as **Adafruit_CircuitPython_74HC165** alongside the existing 75HC595 driver library. If this is of interest, please let me know and I will share my code.

ladyada, 2021-05-18:
hi sure, we accept any/all drivers into the bundle that are `cookiecutter`'d then submit it as a submodule - if you need help with that you can open an issue on the bundle repo :)

closed 2021-05-18 by ladyada

woolseyj, 2021-05-19:
Thank you. I'll do that.