adafruit/circuitpython #4780: f strings don't support slices

Gigahawk, 2021-05-19:
**Firmware**

```python
Adafruit CircuitPython 6.2.0 on 2021-04-05; Adafruit Feather STM32F405 Express with STM32F405RG
```

**Code/REPL**

```python
arr = [1, 2, 3]
print(arr)  # [1, 2, 3]
print(arr[1:])  # [2, 3]
print(f"asdf{arr}")  # asdf[1, 2, 3]
print(f"asdf{arr[1:]}")  # should be asdf[2, 3]
```

**Behavior**

```python
Traceback (most recent call last):
  File "<stdin>", line 1
SyntaxError: invalid syntax
```

Neradoc, 2022-05-21:
This is fixed in latest, thanks to [Micropython 1.18
](https://github.com/micropython/micropython/releases/tag/v1.18)
> lexer: support nested [] and {} characters within f-string params
```py
Adafruit CircuitPython 7.3.0-rc.1 on 2022-05-18; Adafruit CLUE nRF52840 Express with nRF52840
>>> 
paste mode; Ctrl-C to cancel, Ctrl-D to finish
=== arr = [1, 2, 3]
=== print(arr)  # [1, 2, 3]
=== print(arr[1:])  # [2, 3]
=== print(f"asdf{arr}")  # asdf[1, 2, 3]
=== print(f"asdf{arr[1:]}")  # should be asdf[2, 3]
=== 
[1, 2, 3]
[2, 3]
asdf[1, 2, 3]
asdf[2, 3]
>>> 
```

dhalbert, 2022-05-21:
Closing, thanks for checking!

closed 2022-05-21 by dhalbert