adafruit/circuitpython #4781: Ulab Issue compiling CircuitPython only with nRF Ports

bobxor, 2021-05-19:
I'm using an M1 Mac for compiling, following the Markdown pages.  I was successful in compiling the "/ports/atmel-samd" ports, specifically the "circuitplayground_express".

However, when I attempt to compile to any board within "/ports/nrf" I encounter an implicit declaration compiling error with ulab.

```
../../extmod/ulab/code/ndarray.c: In function 'ndarray_object_is_array_like':
../../extmod/ulab/code/ndarray.c:364:8: error: implicit declaration of function 'mp_obj_is_type'; did you mean 'mp_obj_is_true'? [-Werror=implicit-function-declaration]
  364 |     if(mp_obj_is_type(o_in, &ulab_ndarray_type) ||
      |        ^~~~~~~~~~~~~~
      |        mp_obj_is_true
../../extmod/ulab/code/ndarray.c:364:8: warning: nested extern declaration of 'mp_obj_is_type' [-Wnested-externs]
../../extmod/ulab/code/ndarray.c: In function 'generate_slice':
../../extmod/ulab/code/ndarray.c:1156:15: error: implicit declaration of function 'mp_obj_is_int'; did you mean 'mp_obj_list_init'? [-Werror=implicit-function-declaration]
 1156 |     } else if(mp_obj_is_int(index)) {
      |               ^~~~~~~~~~~~~
      |               mp_obj_list_init
../../extmod/ulab/code/ndarray.c:1156:15: warning: nested extern declaration of 'mp_obj_is_int' [-Wnested-externs]
```

I'll take any tips!  Thanks!

tannewt, 2021-05-19:
What does your `git status` show? My guess is that your `ulab` submodule isn't synced up.

bobxor, 2021-05-19:
Thanks for the idea to look at my git.  The issue was seemingly due to switching to the "6.3.x" branch, as I was following this guide:
https://learn.adafruit.com/building-circuitpython/build-circuitpython

I switched back to "main" branch and it now compiles successfully.  Thanks!

closed 2021-05-19 by bobxor