adafruit/circuitpython #4786: i2c is not unlocked with soft reset on 6.2.0

kevinjwalters, 2021-05-20:
On `6.2.0` on a Feather nRF52840 Express using the singleton from `board.I2C()`:

```
Press any key to enter the REPL. Use CTRL-D to reload.

Adafruit CircuitPython 6.2.0 on 2021-04-05; Adafruit Feather nRF52840 Express with nRF52840
>>> import board
>>> i2c = board.I2C()
>>> i2c.try_lock()
True
>>> ["{:2x}".format(x) for x in i2c.scan()]
[' 8', '23', '76']
>>> ["{:02x}".format(x) for x in i2c.scan()]
['23', '76']
>>> ["{:02x}".format(x) for x in i2c.scan()]
['23', '76']
>>>  ### Control-D here then Control-C after that to interrupt
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.

code.py output:
Traceback (most recent call last):
  File "code.py", line 40, in <module>
  File "/lib/pimoroni_pms5003/__init__.py", line 167, in <module>
  File "/lib/pimoroni_pms5003/__init__.py", line 189, in PMS5003
KeyboardInterrupt:

Code done running.

Press any key to enter the REPL. Use CTRL-D to reload.

Adafruit CircuitPython 6.2.0 on 2021-04-05; Adafruit Feather nRF52840 Express with nRF52840
>>> import board
>>> i2c = board.I2C()
>>> i2c.try_lock()
False
>>> i2c.try_lock()
False
>>> i2c.try_lock()
False
>>> i2c.unlock()
>>> i2c.try_lock()
True
>>> ["{:02x}".format(x) for x in i2c.scan()]
['23', '76']
>>> i2c.unlock()
```

Similar theme to https://github.com/adafruit/circuitpython/issues/3266

dhalbert, 2021-05-20:
TThanks! This is a very old bug. It has been true since 5.0.0, probably ever since we started supporting I2C displays. `shared-module/busio/__init__.c` needs to unlock the `board.I2C()` object, as it already does to to the `board.SPI()` object.

dhalbert, 2021-05-20:
> Did I dupe an existing issue? I had a quick look and couldn't find anything obvious.

No, I just mean it has lain undiscovered for quite a while.

kevinjwalters, 2021-05-20:
Unrelated to the issue here, there's a strange address `0x08` (device?) in there which appears on first query and not afterwards. I've got it attached to a Pimoroni Enviro+ FeatherWing which has an LTR-559 (`0x23`) and BME280 (`0x76`) and nothing else AFAIK...

dhalbert, 2021-05-21:
> Unrelated to the issue here, there's a strange address `0x08` (device?) in there which appears on first query and not afterwards. I've got it attached to a Pimoroni Enviro+ FeatherWing which has an LTR-559 (`0x23`) and BME280 (`0x76`) and nothing else AFAIK...

Bring this up in another issue if you think it's a CircuitPython issue. But it might be some I2C device issue.

PR #4801 (2021-05-24): unlock board.I2C() on reset
Fixes #4786.
closed 2021-05-24 by tannewt