adafruit/circuitpython #4803: on the problem of esp32 startup and wake up too long

duanhongyi, 2021-05-22:
Is there any good way to reduce the startup time of circuityython? 
The current deep sleep wake-up takes 4 seconds to enter code.py.

This will consume a lot of power in the actual low-power scenario, and the wake-up time is too long, and the experience is not good.

I tried to use only esp32sdk instead of circuitpython; This time will be reduced to less than one second.

Is there any way？

dhalbert, 2021-05-22:
The current code does not enter true deep sleep if you are connected to a host computer. The idea is to allow you debug a deep-sleep program and not get stuck in an infinite deep sleep loop, where you cannot edit your code, since USB would be disabled during a deep sleep.

However, it can take several seconds for the USB negotiations to finish, so we wait up to 5 seconds to see if USB is really connected.

The issue of being locked out from code changes happened to me personally a number of times while developing the deep sleep code, before I added the delay. I had to erase the board to recover. We thought this was not a good experience. If you have some suggestions about this, let us know.

duanhongyi, 2021-05-22:
Are you talking about waiting for USB? Is there any way to shut it down? Because I don't need USB in mass production, it is only used in development mode.

dhalbert, 2021-05-22:
If you are going to be in mass production, then it might make sense to make your own build of CircuitPython.

The delay is here:
https://github.com/adafruit/circuitpython/blob/20946d97a1be97e07fb9dfc42fa5f07395aa1e54/main.c#L425-L452.

If you want to shorten the delay, you can change the value conditonally defined her by defining in it `ports/esp32/boards/<yourboard>/mpconfigboard.h`.
https://github.com/adafruit/circuitpython/blob/20946d97a1be97e07fb9dfc42fa5f07395aa1e54/py/circuitpy_mpconfig.h#L985-L988

duanhongyi, 2021-05-22:
That's great. Thank you very much.

dhalbert, 2021-05-22:
You're welcome. Closing for now.

closed 2021-05-22 by dhalbert

duanhongyi, 2021-05-22:
Maybe I just need to set `CIRCUITPY_USB_CONNECTED_SLEEP_DELAY` to 0

dhalbert, 2021-05-22:
Yes, try that.

tannewt, 2021-05-24:
> However, it can take several seconds for the USB negotiations to finish, so we wait up to 5 seconds to see if USB is really connected.

Note that the USB sleep shouldn't happen when waking from deep sleep. It should only happen on power on. It's a bug if any delay occurs when waking up.

dhalbert, 2021-05-24:
The delay is happening when `exit_and_deep_sleep()` is called (which then throws an invisible special excepiton), not on wakeup.

tannewt, 2021-05-24:
I don't think it should delay then because ticks should continue counting while sleeping: https://github.com/adafruit/circuitpython/blob/main/main.c#L427

dhalbert, 2021-05-24:
We need to delay the first time because we don't know whether to fake or real deep sleep. @duanhongyi Are you talking about just the first time?

duanhongyi, 2021-05-25:
@dhalbert I think the first time is tolerable.

But I'm doing this now, using a GPIO as the development mode button.

```
import time
import alarm
import board
from digitalio import DigitalInOut, Direction, Pull

_DEBUG = DigitalInOut(board.IO40)
_DEBUG.direction = Direction.INPUT
_DEBUG.pull = Pull.DOWN
if _DEBUG.value:
    time.sleep(10)
else:
    time_alarm = alarm.time.TimeAlarm(monotonic_time=time.monotonic() + 10)
    alarm.exit_and_deep_sleep_until_alarms(time_alarm)
```