adafruit/circuitpython #4807: [Question] About custom external flash.

water5, 2021-05-23:
I try to add new external SPI flash define, on `master` branch, it should be add some define sections likes: 
`#define GD25Q32C {\`
`    .total_size = (1 << 22), /* 4 MiB */ \`
`    .start_up_time_us = 5000, \`
`    .manufacturer_id = 0xc8, \`
`    .memory_type = 0x40, \`
`    .capacity = 0x16, \`
`    .max_clock_speed_mhz = 104, /* if we need 120 then we can turn on high performance mode */ \`
`    .quad_enable_bit_mask = 0x02, \`
`    .has_sector_protection = false, \`
`    .supports_fast_read = true, \`
`    .supports_qspi = true, \`
`    .supports_qspi_writes = true, \`
`    .write_status_register_split = true, \`
`    .single_status_byte = false, \`
`}`

to this file: 
[https://github.com/adafruit/circuitpython/tree/master/supervisor/shared/external_flash/devices.h](https://github.com/adafruit/circuitpython/tree/master/supervisor/shared/external_flash/devices.h)

and modify likes:
`EXTERNAL_FLASH_DEVICES = GD25Q16C`
in
`/boards/[board_name]/mpconfigboard.mk`

But on `main` branch, the `devices.h` droped, below two files instead: 
[https://github.com/adafruit/circuitpython/tree/main/supervisor/shared/external_flash/device.h](https://github.com/adafruit/circuitpython/tree/main/supervisor/shared/external_flash/device.h)
[https://github.com/adafruit/circuitpython/tree/main/supervisor/shared/external_flash/devices.h.jinja](https://github.com/adafruit/circuitpython/tree/main/supervisor/shared/external_flash/devices.h.jinja)

These files not include like below define sections: 
`#define GD25Q32C {\`
`    .total_size = (1 << 22), /* 4 MiB */ \`
`    .start_up_time_us = 5000, \`
`    .manufacturer_id = 0xc8, \`
`    .memory_type = 0x40, \`
`    .capacity = 0x16, \`
`    .max_clock_speed_mhz = 104, /* if we need 120 then we can turn on high performance mode */ \`
`    .quad_enable_bit_mask = 0x02, \`
`    .has_sector_protection = false, \`
`    .supports_fast_read = true, \`
`    .supports_qspi = true, \`
`    .supports_qspi_writes = true, \`
`    .write_status_register_split = true, \`
`    .single_status_byte = false, \`
`}`

I curiously how to explain  `EXTERNAL_FLASH_DEVICES = GD25Q16C` in `main` branch?
and how to add some new SPI flash parameter define?

jposada202020, 2021-05-23:
@water5 I changed to Question and removed Bug is that ok with you?

dhalbert, 2021-05-23:
The `master` branch is defunct. We do develop on `main` now. Also, information about flash chips is now stored in a separate datafile. See 
https://github.com/adafruit/nvm.raw
and
https://github.com/adafruit/nvm.toml

We happy to entertain pull requests to those repos.

closed 2022-02-18 by dhalbert