adafruit/circuitpython #4810: LILYGO TTGO T8 ESP32-S2 w/Display crashes after espidf call

anecdata, 2021-05-26:
**Firmware**

```python
Adafruit CircuitPython 6.2.0 on 2021-04-05; LILYGO TTGO T8 ESP32-S2 w/Display with ESP32S2
```

**Code/REPL**

```python
>>> import gc
>>> gc.mem_free()
8195792
>>> 
>>> import espidf
>>> espidf.heap_caps_get_total_size()
8583106
>>> espidf.heap_caps_get_free_size()
141378
>>> espidf.heap_caps_get_largest_free_block()
```

**Behavior**

The `espidf.heap_caps_get_largest_free_block()` call results in:
```python
Auto-reload is off.
Running in safe mode! Not running saved code.

You are in safe mode: something unanticipated happened.
CircuitPython core code crashed hard. Whoops!
Crash into the HardFault_Handler.
                                 Press any key to enter the REPL. Use CTRL-D to reload.
```

**Description**

I have not seen this on other ESP32-S2 boards (FeatherS2, MagTag, Metro S2, Saola).

`espidf` results differ vastly from FeatherS2, which also has 8MB PSRAM (memory misconfigured?)
```python
Adafruit CircuitPython 6.2.0 on 2021-04-05; FeatherS2 PreRelease with ESP32S2
>>> import gc
>>> gc.mem_free()
8195344
>>> 
>>> import espidf
>>> espidf.heap_caps_get_total_size()
161844
>>> espidf.heap_caps_get_free_size()
130072
>>> espidf.heap_caps_get_largest_free_block()
80896
```

tannewt, 2021-05-27:
> (LILYGO memory misconfigured?)

Does the LILYGO have less memory? That could cause a crash. The UART output from a debug build may have more details too.

anecdata, 2021-05-27:
It should have 8MB, just like FeatherS2. I don't know what these params do, but there are differences in the `sdkconfig` files between the two boards. For example, the LilyGo has these where the FeatherS2 does not:
```
CONFIG_SPIRAM_SPIWP_SD3_PIN=28
# ...
CONFIG_SPIRAM_MALLOC_ALWAYSINTERNAL=16384
CONFIG_SPIRAM_MALLOC_RESERVE_INTERNAL=32768
```
No idea what accounts for `espidf.heap_caps_get_total_size()` for FeatherS2 reporting `161844`, but LilyGo reporting `8583106`.

netroy, 2021-06-09:
not sure if this helps, but I tried getting the UART output, and this is what I got
```
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0x8
load:0x3ffe6108,len:0x1a98
load:0x4004c000,len:0x1f28
load:0x40050000,len:0x1c08
entry 0x4004c3e0
I (46) boot: ESP-IDF v4.3-dev-1197-g8bc19ba89 2nd stage bootloader
I (46) boot: compile time 08:34:44
I (46) boot: chip revision: 0
I (49) qio_mode: Enabling default flash chip QIO
I (54) boot.esp32s2: SPI Speed      : 80MHz
I (59) boot.esp32s2: SPI Mode       : QIO
I (64) boot.esp32s2: SPI Flash Size : 4MB
I (69) boot: Enabling RNG early entropy source...
I (74) boot: Partition Table:
I (78) boot: ## Label            Usage          Type ST Offset   Length
I (85) boot:  0 nvs              WiFi data        01 02 00009000 00005000
I (92) boot:  1 otadata          OTA data         01 00 0000e000 00002000
I (100) boot:  2 ota_0            OTA app          00 10 00010000 00160000
I (107) boot:  3 ota_1            OTA app          00 11 00170000 00160000
I (115) boot:  4 uf2              factory app      00 00 002d0000 00040000
I (122) boot:  5 ffat             Unknown data     01 81 00310000 000f0000
I (130) boot: End of partition table
I (635) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f000020 size=0x3b69c (243356) map
I (679) esp_image: segment 1: paddr=0x0004b6c4 vaddr=0x3ffcdd30 size=0x03218 ( 12824) load
I (682) esp_image: segment 2: paddr=0x0004e8e4 vaddr=0x40028000 size=0x01734 (  5940) load
I (687) esp_image: segment 3: paddr=0x00050020 vaddr=0x40080020 size=0xe87b0 (952240) map
I (867) esp_image: segment 4: paddr=0x001387d8 vaddr=0x40029734 size=0x145fc ( 83452) load
I (887) esp_image: segment 5: paddr=0x0014cddc vaddr=0x40070000 size=0x0001c (    28) load
I (887) esp_image: segment 6: paddr=0x0014ce00 vaddr=0x50000000 size=0x01000 (  4096) load
I (907) boot: Loaded app from partition at offset 0x10000
I (908) boot: Disabling RNG early entropy source...
I (919) cache: Instruction cache        : size 16KB, 4Ways, cache line size 32Byte
I (919) cache: Data cache               : size 16KB, 4Ways, cache line size 32Byte
I (925) spiram: Found 64MBit SPI RAM device
I (930) spiram: SPI RAM mode: sram 40m
I (934) spiram: PSRAM initialized, cache is in normal (1-core) mode.
I (941) cpu_start: Pro cpu up.
I (945) cpu_start: Application information:
I (950) cpu_start: Project name:     circuitpython
I (955) cpu_start: App version:      6.3.0
I (960) cpu_start: Compile time:     Jun  9 2021 12:30:33
I (966) cpu_start: ELF file SHA256:  f63b5d9577235f51...
I (972) cpu_start: ESP-IDF:          v4.2-50-gf30a865fd
I (978) cpu_start: Single core mode
I (2624) spiram: SPI SRAM memory test OK
I (2624) heap_init: Initializing. RAM available for dynamic allocation:
I (2625) heap_init: At 3FF9E01C len 00001FE4 (7 KiB): RTCRAM
I (2631) heap_init: At 3FFDA230 len 00021DD0 (135 KiB): D/IRAM
I (2637) heap_init: At 3FFFC000 len 00003A10 (14 KiB): DRAM
I (2644) cpu_start: Pro cpu start user code
I (2649) spiram: Adding pool of 8192K of external SPI memory to heap allocator
I (2656) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (5512) phy: phy_version: 603, 72dfd77, Jul  7 2020, 19:57:05, 0, 2
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x4002a5d6
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0x8
load:0x3ffe6108,len:0x1a98
load:0x4004c000,len:0x1f28
load:0x40050000,len:0x1c08
entry 0x4004c3e0
E (26) rtc_clk: BBPLL SOFTWARE CAL FAIL
abort() was called at PC 0x4004d7e1
```

to me this wasn't useful at all. perhaps either of you might find something relevant in there.

netroy, 2021-06-09:
sorry, I forgot to do `DEBUG=1` for the build. tried again, and the only relevant bits I found, are these:
```
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x400358ad  PS      : 0x00060b30  A0      : 0x800e4072  A1      : 0x3ffe0cd0
A2      : 0x3f780000  A3      : 0x3ffe0cf0  A4      : 0x0000001c  A5      : 0x3ffe0d4c
A6      : 0x3ffe0d40  A7      : 0x00000001  A8      : 0xfffffffc  A9      : 0x00000000
A10     : 0x3f780010  A11     : 0x00000000  A12     : 0x0000001c  A13     : 0x3ffe0d0c
A14     : 0x3ffe0d00  A15     : 0x00000001  SAR     : 0x00000013  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000000  LBEG    : 0x0000001c  LEND    : 0x3ffe0d0c  LCOUNT  : 0x40029601

Backtrace:0x400358aa:0x3ffe0cd0 0x400e406f:0x3ffe0cf0 0x400e40bd:0x3ffe0d30 0x400aaa99:0x3ffe0d70 0x40093ca3:0x3ffe0d90 0x4008f4ed:0x3ffe0db0 0x4008f5f9:0x3ffe0dd0 0x4009d712:0x3ffe0df0 0x40093e1f:0x3ffe0e90 0x4008f4ed:0x3ffe0ec0 0x4008f51a:0x3ffe0ee0 0x400ae7bd:0x3ffe0f00 0x400aea9e:0x3ffe0fb0 0x400a0a97:0x3ffe0fd0 0x400a0cd2:0x3ffe1000 0x400a0f28:0x3ffe1050 0x400a11f7:0x3ffe1070 0x400df099:0x3ffe1090 0x40031e8d:0x3ffe10c0
```

jepler, 2021-06-09:
> For example, the LilyGo has these where the FeatherS2 does not:
> 
> ```
> CONFIG_SPIRAM_SPIWP_SD3_PIN=28
> # ...
> CONFIG_SPIRAM_MALLOC_ALWAYSINTERNAL=16384
> CONFIG_SPIRAM_MALLOC_RESERVE_INTERNAL=32768
> ```

Based on some source code searching, it appears that these settings do not affect anything.
`SPIRAM_SPIWP_SD3_PIN` applies to ESP32 only, not ESP32-S2.  The `SPIRAM_MALLOC` lines are equal to the defaults.

> I don't know what's changed, but when I try to build this (to look at debug console):
> 
> ```
> 1445232 bytes used,   -3440 bytes free in flash firmware space out of 1441792 bytes (1408.0kB).
> # ...
> Too little flash!!!
> ```

We recently added detection of "too full" flash for esp32s2, which gets you a message like this.  Many or even most boards exceed the available space with DEBUG=1.  Usually, if you flash such an image anyway you get a non-booting board. (#4763)

netroy, 2021-06-09:
I just discovered as well that `CONFIG_SPIRAM_MALLOC_ALWAYSINTERNAL` and `CONFIG_SPIRAM_MALLOC_RESERVE_INTERNAL` were just the default values.

Also, I added `CONFIG_SPIRAM_USE_MEMMAP=y` and that seemed to have fixed the issue.
If this makes sense, I can send a pull-request.

PR #4873 (2021-06-09): Update the sdkconfig for LILYGO TTGO T8 ESP32-S2
* There is no `CONFIG_SPIRAM_SPIWP_SD3_PIN` for [ESP32-S2](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/api-reference/kconfig.html)
* `CONFIG_SPIRAM_MALLOC_ALWAYSINTERNAL` and `CONFIG_SPIRAM_MALLOC_RESERVE_INTERNAL` values are just the defaults. Adding them here is redundant.
* `CONFIG_SPIRAM_USE_MEMMAP=y` fixes #4810
commit 3e463ed8 (2021-06-09): fix #4810
closed 2021-06-09 by jepler