adafruit/circuitpython #4811: Add bits to shift in parameter to GamePadShift

MatzElectronics, 2021-05-26:
I'd like to request that an additional parameter be added to the GamePadShift API/Module that allows you to define the number of bits to be shifted in.  For example, GamePadShift could be used with a Super Nintendo controller if it could be set to shift in 16 bits instead of the default 8:

`classgamepadshift.GamePadShift(clock: digitalio.DigitalInOut, data: digitalio.DigitalInOut, latch: digitalio.DigitalInOut, bits: int = 8)`

I'd be happy to make the changes myself, but the shared bindings are a bit over my head :(

tannewt, 2021-05-26:
Sounds like you found the right place! We're happy to help get you started on discord: https://adafru.it/discord

MatzElectronics, 2021-05-26:
I rebuilt CP and changed shared-module/gamepadshift to verify that this works.  All I changed was the for loop in __init__.c to 16 instead of 8, and all of the relevant uint8_t's in that file and GamePadShift.h to uint16_t's, and that does the trick.  I'm just not 100% about:
 - how to bring that out to the CP API
 - whether making the returned value a uint16_t or even a uint32_t would break any existing code/projects

dhalbert, 2021-05-26:
uint32_t is problematic because without longint support, integers are only 31 bits. So 16 or 24 is better.

dhalbert, 2021-07-27:
`gamepadshift` is now superseded by `keypad.ShiftRegisterKeys`, which lets you specify the number of keys.

closed 2021-07-27 by dhalbert