adafruit/circuitpython #4813: I can’t decode ir remote signal on raspberry pi pico

musimasami, 2021-05-27:
**Firmware**
Adafruit CircuitPython 6.3.0-rc.0 on 2021-05-25; Raspberry Pi Pico with rp2040
Adafruit CircuitPython 6.2.0 on 2021-04-05; Seeeduino XIAO with samd21g18

**Code/REPL**
[codepico.txt](https://github.com/adafruit/circuitpython/files/6552006/codepico.txt)

**REPL**
[3500, 241, 465, 351, 514, 403, 463, 1269, 464, 321, 464, 428, 438, 1268, 465, 429, 437, 402, 252, 404, 462, 1270, 463, 405, 461, 404, 462, 1021, 462, 406, 461, 1272, 462, 405, 86, 1272, 463, 429, 438, 403, 463, 1269, 391, 403, 464, 427, 439, 426, 439, 403, 463, 228, 465, 1269, 466, 1268, 467, 425, 199, 1264, 469, 1242, 493, 425, 117, 424, 440, 425, 441, 1267, 467, 1243, 323, 1268, 467, 1267, 468, 323, 469, 425, 440, 427, 439]

**Behavior**
I tried to decode ir remote signal on raspberry pi pico .
but I some time success to decode ir remote signal . but many times fail.
I also tried to decode ir remote signal on XIAO . successflly done.
How can I resolve this problem ?
I show the data difference between pico and XIAO with code.

**Data**  
**XIAO Seeeduino**
**Pulse width data (reference)**
(usec)
[3543, 1695, 472, 392, 473, 392, 473, 1263, 472, 1263, 471, 393, 472, 1263, 471, 393, 472, 393, 471, 394, 471, 1264, 471, 394, 471, 394, 471, 1265, 470, 394, 471, 1264, 470, 394, 471, 1264, 470, 394, 471, 395, 470, 1265, 469, 395, 470, 396, 469, 395, 470, 395, 470, 1265, 469, 1265, 469, 1265, 469, 395, 470, 1265, 469, 1266, 469, 395, 470, 395, 470, 395, 470, 1266, 468, 1265, 469, 1265, 469, 1265, 469, 1265, 469, 395, 470, 396, 470]
decode result
[44, 82, 9, 55, 62]

**Raspberry pi pico**
**Pulse width data [1]**
(usec)
[3500, 241, 465, 351, 514, 403, 463, 1269, 464, 321, 464, 428, 438, 1268, 465, 429, 437, 402, 252, 404, 462, 1270, 463, 405, 461, 404, 462, 1021, 462, 406, 461, 1272, 462, 405, 86, 1272, 463, 429, 438, 403, 463, 1269, 391, 403, 464, 427, 439, 426, 439, 403, 463, 228, 465, 1269, 466, 1268, 467, 425, 199, 1264, 469, 1242, 493, 425, 117, 424, 440, 425, 441, 1267, 467, 1243, 323, 1268, 467, 1267, 468, 323, 469, 425, 440, 427, 439]

**difference between pico data and XIAO data (XIAO - pico)**
(usec)
43,1454,7,41,-41,-11,10,-6,8,942,7,-35,34,-5,6,-36,35,-9,219,-10,9,-6,8,-11,10,-10,9,244,8,-12,10,-8,8,-11,385,-8,7,-35,33,-8,7,-4,78,-8,6,-31,30,-31,31,-8,7,1037,4,-4,3,-3,2,-30,271,1,0,24,-24,-30,353,-29,30,-30,29,-1,1,22,146,-3,2,-2,1,942,0,-30,30,-31,31

**Pulse width data [2]**
(usec)
[3579, 596, 467, 426, 440, 427, 439, 1268, 466, 677, 467, 427, 439, 1268, 467, 425, 440, 205, 440, 425, 440, 1267, 468, 425, 440, 426, 80, 1268, 467, 400, 466, 1268, 467, 441, 1266, 469, 425, 441, 398, 467, 287, 468, 400, 465, 424, 442, 424, 442, 424, 440, 492, 465, 1270, 464, 1270, 464, 61, 461, 1270, 464, 1271, 464, 405, 461, 332, 463, 429, 437, 1271, 463, 1271, 234, 1271, 465, 1242, 493, 206, 466, 400, 466, 427, 440]

**difference between pico data and XIAO data (XIAO - pico)**
(usec)
-36,1099,5,-34,33,-35,34,-5,6,586,4,-34,33,-5,4,-32,32,188,31,-31,31,-3,3,-31,31,-32,391,-3,3,-6,5,-4,3,-47,-795,795,45,-47,73,-72,183,797,69,-70,46,-46,45,-47,46,-45,-22,800,-801,801,-801,801,408,-66,-800,801,-802,802,64,-66,138,-68,41,-42,-801,803,-803,1031,-802,800,-773,772,263,799,69,-71,43,-44,NaN

**Code**
**pico**
[codepico.txt](https://github.com/adafruit/circuitpython/files/6552010/codepico.txt)

**XIAO**
[codexiao.txt](https://github.com/adafruit/circuitpython/files/6552013/codexiao.txt)

**sensor**
M5STACK IR (SKU:U002)

musimasami, 2021-05-30:
**I added additional Test.**
In Adafruit CircuitPython 7.0.0-alpha.2-697-g96c04b753 on 2021-05-29; Raspberry Pi Pico with rp2040
This version can count 65535usec pulse.This makes it possible to distinguish between noise and signal.

pulse_window=0.15,blocking_delay=0.1(Parameters in Adafruit_irremote)
minimum pulse width = 381usec
83 3115 381 [3115, 1678, 489, 400, 464, 400, 464, 1271, 463, 1270, 458, 408, 456, 1276, 460, 431, 431, 410, 454, 408, 458, 1271, 468, 400, 459, 406, 457, 1276, 463, 400, 464, 1271, 450, 437, 430, 1278, 427, 438, 453, 410, 454, 1279, 459, 381, 485, 406, 456, 408, 454, 413, 450, 1282, 454, 1279, 458, 1254, 480, 407, 458, 1251, 481, 1277, 452, 414, 450, 412, 437, 429, 454, 1279, 453, 1280, 455, 1279, 428, 1306, 457, 1273, 460, 405, 461, 404, 436]
[44, 82, 9, 55, 62]

pulse_window=0.015,blocking_delay=0.01(Parameters in Adafruit_irremote)
minimum pulse width = 134usec
83 3356 134 [3356, 1679, 480, 408, 464, 401, 459, 1275, 464, 1265, 467, 400, 465, 1269, 465, 400, 464, 401, 134, 401, 459, 1275, 463, 396, 468, 372, 486, 1278, 455, 405, 467, 1270, 459, 407, 458, 1274, 462, 398, 467, 400, 460, 1274, 322, 398, 468, 401, 459, 406, 457, 408, 463, 1270, 458, 1275, 461, 1271, 457, 408, 462, 1271, 460, 1269, 467, 402, 461, 402, 321, 368, 496, 1269, 463, 1271, 457, 1251, 483, 1276, 455, 1277, 459, 406, 459, 406, 463]
[44, 82, 9, 55, 62]

I think When pulseio.PulseIn.popleft() is called in ir pulses,short pulse width occurs.
More better!

tannewt, 2021-06-01:
@musimasami Is it fixed in absolute latest then? We'll get a proper alpha out this week.

musimasami, 2021-06-02:
I found that the error was reduced by extending the pulse_window time. But I will miss that multiple pulses are generated by pressing and holding the button.
The time of one ir remote data of the air conditioner is about 0.24 seconds.
**Panasonic Air conditioner decode results**
**Adafruit CircuitPython 6.2.0 on 2021-04-05; Seeeduino XIAO with samd21g18**
pulse_window=0.04 decode result
[2, 32, 224, 4, 0, 0, 0, 6, 2, 32, 224, 4, 0, 64, 44, 128, 175, 0, 0, 6, 96, 0, 0, 128, 0, 6, 141]
**Adafruit CircuitPython 7.0.0-alpha.2-697-g96c04b753 on 2021-05-29; Raspberry Pi Pico with rp2040**
**pulse_window=0.04 decode result**
[2, 32, 224, 4, 0, 0, 0, 6, 2, 32, 224, 4, 0, 64, 44, 128, 175, 0, 0, 6, 96, 0, 0, 0, 0, 6, 141] >> Error
**pulse_window=0.104 decode result**
[2, 32, 224, 4, 0, 0, 0, 6, 2, 32, 224, 0, 0, 64, 44, 128, 175, 0, 0, 6, 96, 0, 0, 128, 0, 6, 141] >> Error
[2, 32, 224, 4, 0, 0, 0, 6, 2, 32, 224, 0, 0, 64, 44, 128, 175, 0, 0, 6, 96, 0, 0, 128, 0, 6, 141] >> Error
**pulse_window=0.1504 decode result**
[2, 32, 224, 4, 0, 0, 0, 6, 2, 32, 224, 4, 0, 64, 44, 128, 175, 0, 0, 6, 96, 0, 0, 128, 0, 6, 141] >> No error

**Ir remote signal detail**
**Panasonic Air conditioner**
**pulses sum: 238081usec** 
pulse count: 439 
pulses: [3217, 1709, 466, 403, 459, 1253, 485, 408, 458, 405, 469, 404, 465, 402, 461, 408, 460, 408, 457, 406, 468, 405, 465, 401, 459, 408, 467, 403, 458, 1276, 468, 402, 463, 402, 467, 407, 463, 399, 468, 403, 460, 405, 468, 404, 463, 1251, 480, 1274, 469, 1273, 456, 408, 466, 404, 459, 1276, 466, 404, 460, 405, 467, 403, 459, 410, 457, 386, 488, 405, 463, 398, 471, 404, 461, 407, 54, 409, 457, 405, 468, 403, 460, 408, 466, 403, 461, 408, 460, 403, 469, 403, 459, 410, 464, 401, 465, 406, 457, 406, 468, 404, 457, 412, 464, 398, 467, 407, 459, 405, 468, 402, 461, 410, 464, 397, 470, 403, 458, 1279, 465, 1271, 459, 411, 461, 401, 468, 403, 460, 405, 468, 403, 459, 9962, 3491, 1687, 343, 408, 459, 1271, 470, 406, 457, 409, 465, 402, 461, 409, 462, 400, 470, 403, 460, 405, 468, 403, 459, 410, 463, 401, 467, 403, 459, 1254, 489, 402, 461, 408, 464, 399, 470, 405, 457, 410, 464, 399, 468, 406, 457, 1251, 490, 1273, 460, 1250, 490, 403, 460, 410, 462, 1248, 488, 404, 459, 408, 467, 403, 462, 409, 455, 407, 467, 405, 463, 404, 463, 402, 468, 403, 460, 409, 463, 400, 469, 403, 459, 410, 460, 404, 468, 399, 469, 399, 385, 404, 459, 406, 467, 398, 468, 1275, 462, 399, 469, 404, 460, 410, 462, 1269, 468, 1271, 466, 403, 461, 1276, 464, 400, 467, 404, 460, 409, 463, 402, 467, 403, 461, 408, 461, 402, 469, 402, 460, 411, 463, 1268, 468, 1273, 459, 1271, 469, 1272, 457, 1279, 464, 403, 465, 1273, 462, 400, 470, 1271, 459, 411, 464, 402, 465, 404, 458, 405, 468, 404, 459, 410, 464, 399, 469, 403, 462, 407, 459, 311, 468, 403, 462, 404, 467, 402, 462, 405, 467, 403, 464, 404, 464, 406, 457, 1277, 466, 1267, 468, 404, 459, 410, 464, 400, 467, 404, 459, 406, 468, 403, 460, 408, 462, 403, 468, 402, 461, 410, 457, 1254, 487, 1248, 483, 411, 458, 405, 469, 403, 458, 407, 467, 404, 462, 405, 459, 406, 469, 402, 464, 405, 459, 407, 463, 404, 468, 404, 464, 402, 460, 406, 466, 406, 465, 402, 457, 409, 467, 403, 464, 404, 459, 408, 465, 398, 470, 404, 464, 405, 458, 411, 462, 323, 469, 404, 459, 406, 468, 403, 462, 408, 456, 406, 468, 402, 467, 404, 464, 400, 468, 404, 457, 1253, 482, 1280, 459, 409, 465, 402, 459, 385, 486, 404, 468, 403, 465, 1268, 468, 404, 465, 1272, 464, 1270, 465, 404, 458, 404, 469, 406, 461, 1273, 457]

**Illumination Ir remocon**
**pulses sum: 53619** 
pulse count: 83 
pulses: [3059, 1718, 468, 403, 463, 409, 464, 1279, 464, 1277, 470, 403, 466, 1254, 485, 411, 469, 401, 464, 410, 468, 1275, 461, 407, 472, 400, 463, 1257, 493, 404, 464, 1254, 488, 410, 467, 1275, 462, 406, 472, 402, 465, 1280, 464, 410, 466, 402, 463, 411, 466, 309, 557, 1279, 470, 1275, 464, 1255, 487, 1276, 469, 409, 460, 848, 467, 401, 467, 409, 467, 402, 464, 1281, 461, 1254, 490, 405, 473, 404, 467, 1275, 464, 382, 495, 402, 467]

musimasami, 2021-06-02:
Improved. But No.
I uploaded comments and detailed datas.

musimasami, 2021-06-10:
I tested 
**Adafruit CircuitPython 7.0.0-alpha.3 on 2021-06-03; Raspberry Pi Pico with rp2040**

It looks fine.
I can detect multiple pulses without error.

**Data**
**Illumination Ir remocon**
sum: 55436 pulse count: 83 pulses: [3141, 1708, 459, 409, 483, 377, 492, 1275, 464, 1269, 467, 392, 438, 1269, 456, 441, 467, 397, 468, 391, 460, 1274, 461, 403, 463, 403, 460, 1278, 463, 395, 464, 1249, 458, 403, 457, 1275, 459, 434, 432, 408, 487, 1275, 464, 394, 463, 406, 464, 398, 455, 383, 493, 1272, 454, 1279, 454, 1273, 433, 433, 458, 1274, 460, 1153, 491, 401, 457, 405, 462, 409, 459, 1241, 493, 1242, 489, 1275, 457, 1272, 435, 1276, 483, 380, 487, 402, 433]
 [44, 82, 9, 55, 62]
pulses sum: 55562 pulse count: 83 pulses: [3527, 1678, 494, 393, 434, 440, 463, 1268, 460, 980, 434, 408, 490, 1270, 466, 394, 434, 434, 461, 405, 464, 1244, 483, 401, 433, 438, 460, 1246, 454, 411, 484, 1276, 460, 404, 455, 1278, 464, 399, 454, 384, 487, 1272, 459, 407, 466, 397, 466, 394, 442, 424, 462, 1249, 455, 1297, 433, 1302, 431, 433, 463, 1247, 485, 1276, 456, 408, 463, 397, 465, 404, 434, 1300, 455, 1252, 487, 1269, 397, 1274, 432, 1283, 483, 404, 463, 402, 462]
[44, 82, 9, 55, 62]
pulses sum: 55562 pulse count: 83 pulses: [3496, 1703, 490, 407, 435, 430, 455, 1251, 489, 1267, 434, 435, 462, 1266, 434, 434, 432, 437, 465, 394, 435, 1271, 490, 407, 97, 380, 460, 1301, 432, 430, 436, 1275, 485, 408, 462, 1272, 457, 409, 465, 395, 466, 1268, 435, 431, 427, 417, 485, 401, 434, 404, 484, 1279, 459, 1275, 467, 1235, 455, 415, 485, 1269, 461, 1245, 465, 432, 461, 399, 429, 437, 438, 1295, 435, 1303, 454, 1280, 467, 1236, 454, 1305, 433, 430, 435, 438, 456]
 [44, 82, 9, 55, 62]

**Panasonic Air conditioner**
pulses sum: 243913 pulse count: 439 pulses: [3298, 1703, 462, 413, 464, 1244, 491, 399, 461, 413, 455, 413, 464, 403, 465, 399, 463, 410, 456, 412, 464, 404, 464, 400, 467, 400, 460, 386, 489, 1275, 463, 404, 464, 398, 466, 409, 458, 410, 463, 404, 464, 399, 467, 406, 458, 1250, 492, 1268, 461, 1279, 456, 412, 463, 404, 465, 1272, 458, 410, 461, 403, 465, 400, 464, 411, 459, 375, 496, 405, 461, 406, 458, 412, 460, 408, 464, 278, 461, 414, 464, 402, 466, 397, 462, 413, 459, 379, 492, 406, 464, 402, 457, 413, 461, 406, 465, 403, 464, 396, 465, 412, 464, 405, 456, 412, 462, 403, 466, 394, 466, 411, 461, 381, 490, 403, 458, 412, 463, 1238, 498, 1247, 482, 412, 457, 376, 498, 400, 467, 398, 463, 412, 458, 9959, 3496, 1682, 264, 401, 468, 1271, 455, 414, 464, 399, 467, 399, 463, 377, 498, 401, 464, 407, 457, 412, 458, 377, 495, 401, 467, 404, 458, 385, 490, 1272, 456, 413, 463, 404, 464, 404, 455, 414, 464, 398, 466, 408, 461, 407, 464, 1271, 457, 1280, 464, 1246, 488, 399, 461, 414, 464, 1272, 464, 397, 464, 411, 456, 382, 494, 400, 467, 399, 460, 414, 462, 373, 496, 396, 463, 414, 463, 405, 464, 400, 464, 409, 454, 413, 462, 379, 491, 399, 464, 409, 204, 406, 463, 1265, 462, 1282, 455, 413, 463, 398, 464, 404, 460, 1282, 460, 1276, 464, 397, 463, 1280, 465, 1266, 462, 412, 463, 400, 461, 411, 465, 395, 463, 381, 495, 406, 462, 402, 462, 409, 465, 396, 466, 1276, 460, 1276, 457, 1273, 464, 1278, 465, 1273, 461, 407, 463, 1273, 455, 413, 463, 1266, 463, 383, 491, 401, 468, 404, 463, 406, 463, 397, 463, 412, 462, 407, 458, 358, 463, 399, 462, 412, 461, 404, 466, 398, 461, 414, 457, 410, 464, 396, 466, 412, 462, 399, 468, 1269, 460, 1280, 462, 404, 463, 408, 458, 410, 463, 401, 460, 413, 462, 398, 461, 380, 496, 406, 458, 410, 461, 408, 462, 1273, 458, 1251, 482, 381, 489, 411, 463, 399, 463, 410, 464, 404, 461, 382, 483, 411, 459, 1270, 462, 413, 463, 400, 463, 410, 461, 379, 489, 406, 462, 407, 463, 400, 460, 413, 463, 404, 456, 384, 492, 399, 470, 398, 463, 411, 461, 408, 462, 47, 353, 405, 454, 1282, 463, 395, 465, 413, 463, 401, 462, 410, 463, 401, 458, 380, 496, 406, 463, 398, 465, 379, 493, 1275, 462, 1272, 461, 407, 464, 1273, 464, 398, 464, 412, 462, 402, 465, 1273, 457, 1276, 467, 1272, 462, 407, 463, 1272, 460, 379, 492, 1266, 467, 1276, 463]
T: [2, 32, 224, 4, 0, 0, 0, 6, 2, 32, 224, 4, 0, 48, 54, 128, 175, 0, 0, 6, 96, 64, 0, 128, 0, 22, 215]

musimasami, 2021-06-16:
I want to close this issue.

closed 2021-06-16 by musimasami