adafruit/circuitpython #4814: Add ulab to KittenBot Meowbit (STM32F401RET6)

kevinjwalters, 2021-05-27:
I was surprised to find ulab absent from the KittenBot Meowbit recently as the devices for MakeCode Arcade generally have a decent amount of resources like flash.

Discussed this in the comments in https://github.com/adafruit/circuitpython/pull/2693#issuecomment-841891259 and @hierophect mentioned it's more to do with the fiddly aspects of layout then quantity and it could be done.

kevinjwalters, 2021-05-27:
I don't have immedite need for this but it would be nice to "port" the code from my new [component tester](https://www.youtube.com/watch?v=umDHGAauNhs) onto it.

ladyada, 2021-05-27:
interesting! this isnt something we have any plans on working on - @kevinjwalters please try! any PR is welcome!

hierophect, 2021-05-27:
lazy copy paste, here's how Micropython packs structs into the ISR section.
```
    /* The startup code goes first into FLASH */
    .isr_vector :
    {
        . = ALIGN(4);
        KEEP(*(.isr_vector)) /* Startup code */

        /* This first flash block is 16K annd the isr vectors only take up
           about 400 bytes. So we pull in a couple of object files to pad it
           out. */

        . = ALIGN(4);

        /* NOTE: If you update the list of files contained in .isr_vector,
                 then be sure to also update smhal/Makefile where it forcibly
           builds each of these files with -Os */

        */ff.o(.text*)
        */vfs_fat_*.o(.text*)
        */py/formatfloat.o(.text*)
        */py/parsenum.o(.text*)
        */py/mpprint.o(.text*)

        . = ALIGN(4);
    } >FLASH_ISR
```

Probably just a bit of trial and error to figure out what fits in the Meowbit's case and testing it to make sure nothing breaks.