adafruit/circuitpython #4817: Cannot use ctrl-C to interrupt code.py after sending data to the REPL

Neradoc, 2021-05-27:
When code.py is running, sending a certain quantity of characters to the serial port disables ctrl-C.
Here is a test python script that takes the serial as argument, and can be run to see if it interrupts the code, using a simple `code.py` blinking an LED. Different values of `NUMCHARS` will cause the issue depending on the board/port.
```py
# that's for convenience
import argparse
parser = argparse.ArgumentParser()
parser.add_argument('port', type=str, help="Serial port of the board", nargs=1)
args = parser.parse_args()
serial_port = args.port[0]

# that's the actual test
import serial
import time

CTRLC = b'\x03'
NUMCHARS = 1000

channel = serial.Serial(serial_port)
channel.write(b" " * NUMCHARS)
channel.write(CTRLC)
channel.close()
```
That was discovered when using Mu, since clicking in the serial panel makes it send characters to move the insertion point in the REPL, leading to the issue. https://github.com/mu-editor/mu/issues/1505

(Tested on MacOS Catalina with the circuitplayground and a few different boards)

tannewt, 2021-06-01:
I'm not sure this is a bug because the spaces are getting queued up by TinyUSB and not being read by the user code. The CTRL-C then is never actually transmitted to the device because the USB buffers are full.

Neradoc, 2021-06-03:
Oh, so it would require to modify TinyUSB if we want to be able to intercept ctrl-C with a full buffer ?
If we can't solve it, then maybe the Mu issue should be fixed over there, since I found the issue really easy to trigger with a Circuit Playground Express for example.

dhalbert, 2021-06-03:
There is code to watch for a wanted character:
https://github.com/adafruit/circuitpython/blob/26751c9afe612b1b56db1cbf24651f76f93215d1/supervisor/shared/usb/usb.c#L81-L88
which causes this callback:
https://github.com/adafruit/circuitpython/blob/26751c9afe612b1b56db1cbf24651f76f93215d1/supervisor/shared/usb/usb.c#L290-L298

So it does sound like the ctrl-c has not arrived yet.

Tagging @hathach for interest.

tannewt, 2021-06-03:
> Oh, so it would require to modify TinyUSB if we want to be able to intercept ctrl-C with a full buffer ?
> If we can't solve it, then maybe the Mu issue should be fixed over there, since I found the issue really easy to trigger with a Circuit Playground Express for example.

What is the state of circuitpython when this happens? If nothing is reading the serial then I'm not sure the ctrl-c is ever actually sent from the host. TinyUSB won't accept more CDC data if it's buffers are full.

dhalbert, 2021-07-15:
Since the buffers are full, we never see the ctrl-C, because it hasn't been delivered yet. So we cannot fix this on the CircuitPython side.

closed 2021-07-15 by dhalbert