adafruit/circuitpython #4818: Pasting into the REPL can skip characters on NRF

Neradoc, 2021-05-27:
On NRF52 boards (CLUE, Feather NRF52840 Express) pasting code to the REPL can lead to characters missing sometimes, unpredictably. I couldn't reproduce the issue on Circuit Playground Express or other non NRF boards.

For example repeatedly pasting `"| | | | | | | | | | | | | | | | | | | | | | | | |"` (with a return) can lead to that kind of display (can take multiple tries to reproduce).
```
>>> "| | | | | | | | | | | | | | | | | | | | | | | | |"
'| | | | | | | | | | | | | | | | | | | | | | | | |'
>>> "| |  | | | | | | | | | | | | | | | | | | | | | |"
'| |  | | | | | | | | | | | | | | | | | | | | | |'
>>> "| | | | | | | | | | | | | | | | | | | | | | | | |"
'| | | | | | | | | | | | | | | | | | | | | | | | |'
>>> "| | | | | | | | | | | | | | | | | | | | | | | | |"
'| | | | | | | | | | | | | | | | | | | | | | | | |'
>>> "| | | | || | | | | | | | | | | | | | | | | | | |"
'| | | | || | | | | | | | | | | | | | | | | | | |'
>>> "| |  | | | | | | | || | | | | | | | | | | | | | |"
'| |  | | | | | | | || | | | | | | | | | | | | | |'
```

lesamouraipourpre, 2021-05-28:
I think @FoamyGuy had something similar on one of their streams. Pasting into the REPL wasn't reliable. I can't remember what the board was.

lesamouraipourpre, 2021-05-28:
I found the stream:
https://youtu.be/fkEcwVXkMgc?t=2049

The first bad paste is at 34:24

My gut feeling is that there is a (keyboard) buffer that is overflowing, but why is it only losing a single character at a time?

DavePutz, 2021-05-28:
Please check the comments in issue #3152. What I found was that the flaky pasting depended upon how the client program was sending the string to the CP board.

Neradoc, 2021-05-28:
Oh it's a known issue, I completely missed it !
Well sorry for that.

closed 2021-05-28 by Neradoc