adafruit/circuitpython #4821: Play specific part of a WaveFile

rsbohn12, 2021-05-28:
I want to take a small slice from a larger .wav file and loop it.

Something like this would be great:

    dac.playSlice(waveFile, offset, duration)

Units for offset and duration would be samples, the user would have to be aware of the sample rate.

I want to build a handheld audio sample and playback device. The current workflow is to cut up the samples using Audacity, export individual slices as .wav files, then play them on the device.

tannewt, 2021-06-01:
Another way to do it would be to add reading of samples from the Wave and then they could be loaded into their own RawSample to be looped.

Monsieur36, 2022-11-02:
Might properties such as start point and length/duration of the wav to be played be accessible via audiocore/audioio/audiomixer now or in future?