adafruit/circuitpython #4822: HTML parsing capabilities/library

Dynamonic, 2021-05-28:
A lot of webpages return HTML responses instead of JSON objects. It would be great if there were a simple HTML parser built into circuitpython that could allow users to extract data from HTML response webqueries for their projects.

dhalbert, 2021-05-28:
A lightweight Python-based library might make sense, though "HTML" and "lightweight parsing" don't tend to go together. I don't think we'd have the room for a C-based one in a number of boards. Using regular expressions to scrape data from HTML is frowned upon, but is possible. Are you using `re` right now?

Dynamonic, 2021-06-01:
I tried to look into re type stuff but could not get it to work. I think I just need to get more familiar with this.

tannewt, 2021-06-01:
The first place to start when wanting to add an API is to look at what is available in CPython. I think the most common Python version is BeautifulSoup: https://www.crummy.com/software/BeautifulSoup/bs4/doc/

Then the challenge is picking the small subset to provide in CP.