adafruit/circuitpython #4828: Nano Connect RP2040 Requests Error

dedSyn4ps3, 2021-05-29:
**Firmware**

```python
Adafruit CircuitPython 6.2.0-beta.2 on 2021-05-28; Arduino Nano Connect RP2040
Adafruit CircuitPython 7 built on 2021-05-27; Arduino Nano Connect RP2040
```

**Code/REPL**

```python
--- snip ---
print("Connected to", str(esp.ssid, "utf-8"), "\tRSSI:", esp.rssi)
print("My IP address is", esp.pretty_ip(esp.ip_address))

print()
print("Fetching json from Initial State")
r = requests.get("https://api.init.st/data/v1/events/latest?accessKey=abcd1234&bucketKey=efg5678") # This is line 52
print(r.text)
r.close()
--- snip ---
```

**Behavior**

```python
Firmware vers. bytearray(b'1.4.5\x00')
MAC addr: ['0x6c', '0xb9', '0x2f', '0xa8', '0xcc', '0x84']
Connecting to AP...
Connected to GL-AR-Guest 	RSSI: -64
My IP address is 192.168.9.222

Fetching json from Initial State
Traceback (most recent call last):
  File "code.py", line 52, in <module>
  File "adafruit_requests.py", line 684, in get
  File "adafruit_requests.py", line 562, in request
  File "adafruit_requests.py", line 429, in _get_socket
RuntimeError: Sending request failed
```

**Description**

- Error establishing socket to make request...
- Only happens when making a request (so far, haven't tested additional cases)...
- uBlox Co-processor makes successful connection to AP, so no issue there..only when attempting to make a request...

**Additional Info**

Prior to the official release on the Circuitpython.org website for the Arduino Nano Connect RP2040, I had compiled the "Absolute Newest" testing branch for the board right after the pull request was made and added it. I compiled everything and flashed it onto the nano connect.

I2C, SPI, and all the other basic protocols worked just fine, and I noticed that the release version was for circuitpython 7, so I went ahead and downloaded both the compiled .mpy bundle, as well as the .py bundle to test making web requests with both. Each time the board would successfully connect to wifi, retrieve an IP Address via DHCP...but then would fail upon trying to establish a proper socket.

The next day, I saw the official release for the board on circuitpython.org, but this time using the version 6.x of the firmware. I downloaded that, re-flashed my board, and then downloaded the bundles for the 6.x version of the firmware. Once again, there is a failure to establish a proper socket when attempting to make a request using the 'adafruit_requests' library.

I have not had the time to dig into the library code to locate what exactly seems to be the issue, but am attaching a screen snip of the debugger lines showing where in the code the problem seems to stem from. I realize that this board is super new, and development is ongoing. I'll keep tinkering with it, but also have issues using the C/C++ side of things as well with the MQTT library for Adafruit IO (which I have made a post about on the Adafruit Forums).

[image]

ladyada, 2021-05-29:
what if you use the URL / pinout / code from
https://github.com/adafruit/circuitpython/pull/4802
(without the OLED display)

dedSyn4ps3, 2021-05-29:
Thanks for the super fast response!

That was exactly where I pulled the pinout from for my code, was from your wifi example. I first started with that exact code, minus the OLED, to retrieve the DOGE coin value. The URL gave me the same error, so I thought I would try and use a different URL and pull the sensor values from my Initial State dashboard like I do with several other boards I have.

All of them keep giving a socket error whenever trying to make the request..

dedSyn4ps3, 2021-05-30:
@ladyada , I've narrowed down the requests issue. It definitely seems to be an SSL/TLS issue with the request and socket establishment. I ran a small webserver on my phone that I've used in the past to serve code files up real fast, and it uses basic HTTP. The Nano Connect had no issues retrieving the contents from it!
[image]

dedSyn4ps3, 2021-05-30:
### @ladyada , I have also just tested and noticed the exact same behavior using MQTT. I adjusted the RP2040 pubsub code for Adafruit IO to use the proper pinout for the Nano Connect, and once again experienced failure trying to connect to AIO using MQTTS port 8883.

**As soon as I manually added the standard port 1883 when creating the mqtt_client object, all works fine once again! Definitely something up when trying to use TLS...**

[image]

dedSyn4ps3, 2021-06-11:
**@tannewt @ladyada** ,

I have been emailing back and forth with Arduino about the TLS issues experienced between both languages. The issue seems to somewhat be corrected as of right now. The WifiSSLClient sketch that I run is now able to connect and retrieve the site data for a google search query over HTTPS. The oddity is that in the process of getting the code to work, I switched the access point I was using which no longer used '8.8.8.8' and '8.8.4.4' as it's DNS servers, and started having success running the sketch.

I hopped over to circuitpython on the board and re-ran the DOGE coin example code (which uses an https url), and it works! I got super excited so I changed up the url to use the Initial State API to query sensor data from my dashboard, and you guessed it...socket error just like in my screenshot above. I have even tried using the basic Adafruit IO API query with my key included as a header....and it too still fails.

One last thing I tried to do was load up the MQTT code displayed above, but using secure port 8883....fails still :'(

[image]

tannewt, 2021-06-22:
@brentru Is this the IO cert issue in the nina FW?

brentru, 2021-06-22:
@tannewt  Yes, I believe this is an issue with Arduino's nina-fw upstream not containing the root cert. for Adafruit IO, since @dedSyn4ps3 's sketch works properly w/insecure port 1883 (non-SSL connection).

dedSyn4ps3, 2021-06-22:
**@brentru** I sent an email to arduino last week in regards to adding the root cert, but have not heard back. I'm not sure if this is because they closed the issue ticket that I originally filed for the https issue...I may have to reopen a new ticket, not sure.

I know that from prior experience using the Seeed Wio Terminal in the Arduino IDE, I would have to manually add the root cert in my sketch for proper HTTPS comms to whatever site I was trying to communicate with. I have not had the time between work and mobile app dev to really look into the circuitpy docs, but is there a temporary solution to include either a byte array or dictionary variable holding the certificate info, then use that with the requests client? I thinking along the lines of how Golang programs I've written load a root CA for client/server comms...

anecdata, 2021-06-22:
Can the Adafruit fork of NINA be flashed onto the Nano (are the U-blox® pins the same)?

brentru, 2021-06-22:
@anecdata  I do not believe so, they've done some upstream PRs specifically for the RP2040 Connect.

ngdrascal, 2021-08-30:
I was experiencing this exact problem.  Some HTTPS websites worked fine and others issued a traceback originating in _get_socket.  After going to Arduino's IO Cloud website (create.arduino.cc/iot) and updating the firmware for the u-blox the error went away.  The device came with NINA firmware version 1.4.5.  The upgraded version is 1.4.8.

dedSyn4ps3, 2021-08-30:
> I was experiencing this exact problem.  Some HTTPS websites worked fine and others issued a traceback originating in _get_socket.  After going to Arduino's IO Cloud website (create.arduino.cc/iot) [...]

Intersting @ngdrascal ... when I get a chance I will have to check that out. Have you been able to recreate both the HTTPS and MQTTS code examples from above using or adafruit example codes since updating the ublox firmware?

hjubb, 2021-08-31:
I haven't been using MQTT/S but was having issues regarding HTTPS encountering the exact same `_get_socket` traceback in an error. My solution was to install the root certs for the specific domains *I was targeting* (which wasn't made very clear from any of the examples I could find) using the arduino-fwuploader at https://github.com/arduino/arduino-fwuploader

ngdrascal, 2021-08-31:
> Intersting @ngdrascal ... when I get a chance I will have to check that out. Have you been able to recreate both the HTTPS and MQTTS code examples from above using or adafruit example codes since updating the ublox firmware?

@dedSyn4ps3 The Adafruit example from https://learn.adafruit.com/circuitpython-on-the-arduino-nano-rp2040-connect/wifi actually worked fine before I upgraded the ublox firmware.  I wasn't experiencing the issue with all websites, just some.  For example posting to https://reqbin.com/echo/post/form works fine, but posting to https://login.microsoftonline.com failed in the __get_socket_ routine.  After upgrading the ublox firmware posting to https://login.microsoftonline.com worked fine.

I have not tried the MQTT example.

dedSyn4ps3, 2021-08-31:
> > Intersting @ngdrascal ... when I get a chance I will have to check that out. Have you been able to recreate both the HTTPS and MQTTS code examples from above using or adafruit example codes since updating the ublox firmware?
> 
> @dedSyn4ps3 The Adafruit example from https://learn.adafruit.com/circuitpython-on-the-arduino-nano-rp2040-connect/wifi actually worked fine before I upgraded the ublox firmware. I wasn't experiencing the issue with all websites, just some. For example posting to https://reqbin.com/echo/post/form works fine, but posting to https://login.microsoftonline.com failed in the __get_socket_ routine. After upgrading the ublox firmware posting to https://login.microsoftonline.com worked fine.
> 
> ~I have not tried the MQTT example.~
> 
> I was able to run the MQTT demo (https://learn.adafruit.com/mqtt-in-circuitpython/overview) **without any issues** on the Arduino Nano rp2040. If I understand correctly port 8883 uses SSL.
> 
> ```
> mqtt_client = MQTT.MQTT(
>     broker="io.adafruit.com",
>     username=secrets["aio_username"],
>     password=secrets["aio_key"],
>     port=8883
> )
> ```

@ngdrascal fantastic...so you're saying that the snip above is what you ran with no errors? I just hopped on to post my own success on the arduino-side of things when I saw your update. 

@ladyada, I updated the uBlox firmware similar to @ngdrascal just through the IDE itself. I'm running the 2.0 beta on macOS. After the fw updated I added the root ca for adafruit as well as initial state, which is what my example code successfully connects and posts data to using TLS.

[image]

[image]

### Maybe an update to the Nano Connect guide could include some lovely shout-outs this time around from us contribs ;)

ngdrascal, 2021-08-31:
> @ngdrascal fantastic...so you're saying that the snip above is what you ran with no errors? I just hopped on to post my own success on the arduino-side of things when I saw your update.

@dedSyn4ps3 Correct.

In my case I DID NOT upload any certs.  I just updated firmware on the ublox.

dedSyn4ps3, 2021-08-31:
@ngdrascal gotcha. The reason why I uploaded the certs was due to a 403 Forbidden when first attempting to make the API request for Initial State.

After uploading the root cert and double checking my request query everything went through just fine. 🤘👌

dedSyn4ps3, 2021-08-31:
@tannewt , I think we've just about got this bug squashed...I'll double check the CP version of querying my initial state dashboard to see if it goes through after the ublox update.

If it doesn't,  I have not found anywhere in the docs of a way to include a root cert or .pem file into CP code like can be done through arduino. Any way to do that if there still is an issue with some endpoints?

dedSyn4ps3, 2021-08-31:
> @ngdrascal gotcha. The reason why I uploaded the certs was due to a 403 Forbidden when first attempting to make the API request for Initial State.
> 
> After uploading the root cert and double checking my request query everything went through just fine. 🤘👌

@ngdrascal @ladyada , I re-loaded CP onto the board and TLS queries to the Initial State API are now functional as well...great success!

I don't know if the ublox update alone would have fixed this, since the arduino request generated a 403 response prior to adding the root cert for initial state.

If users find that a TLS endpoint is not working, they may just need to pull up the arduino IDE and add the cert for where they are communicating with first 👍🤘😎

tannewt, 2021-08-31:
> @tannewt , I think we've just about got this bug squashed...I'll double check the CP version of querying my initial state dashboard to see if it goes through after the ublox update.

Great! Will close it now.

> If it doesn't, I have not found anywhere in the docs of a way to include a root cert or .pem file into CP code like can be done through arduino. Any way to do that if there still is an issue with some endpoints?

I don't know of one. I suspect we'd need to add the cert loading in the [ESP32SPI library](https://github.com/adafruit/Adafruit_CircuitPython_ESP32SPI) though since the ESP does all of the heavy lifting.

closed 2021-08-31 by tannewt