adafruit/circuitpython #4830: PID for nanoESP32-S2 w/WROVER

reppad, 2021-05-30:
Hi Adafruit !

I would like to use CircuitPython on my MuseLab nanoESP32-S2 WROVER board, so I started to create the board definition - based on the nanoESP32-S2 WROOM board - to submit a PR but realized that the WROOM version uses Adafruit USB VID 0x239A.
It would probably be more consistent if the WROVER version also uses this VID, is it possible to assign one PIDs to this board?

The branch with the new board (and the reword of WROOM version to be more explicit)
https://github.com/reppad/circuitpython/tree/muselab_nanoesp32_s2_wrover

Thanks !
Reppad

ladyada, 2021-05-30:
hi, espressif can assign you a PID, that way it has the espressif VID. you'll need two per board as one is needed for the bootloader. please check with them - :)

reppad, 2021-05-30:
Thanks, I'll try to ask Espressif :)
So the WROOM version that already exists [have an Adafruit VID](https://github.com/adafruit/circuitpython/blob/main/ports/esp32s2/boards/muselab_nanoesp32_s2/mpconfigboard.mk) and the WROVER version of the same board will have an Espressif VID, right ?

ladyada, 2021-05-30:
right! might be a good idea to get matching ones - altho really thats up to you, it doesnt matter as we have many VID's

reppad, 2021-05-31:
Espressif PIDs obtained :)
```
0x80B2 | Muse Lab nanoESP32-S2 WROVER - CircuitPython  
0x80B3 | Muse Lab nanoESP32-S2 WROVER - UF2 Bootloader
```
Thanks @ladyada !

closed 2021-05-31 by reppad