adafruit/circuitpython #4831: Allow displayio to use BytesIO for bitmap display

wifijt, 2021-05-30:
Current examples that download images from the web require you to write them to storage and then display them. This requires the filesystem to be read-write , a potentially destructive requirement. 

Circuitpython io supports BytesIO, but there is no way to display an image that's loaded into a BytesIO object. 

bg_bitmap_file = BytesIO(requests.get(url, stream=True).content)
bg_bitmap = displayio.OnDiskBitmap(bg_bitmap_file)
This code doesn't work as is throws an object type error.

I propose a new method displayio.BytesIOBitmap(bg_bitmap_file) that would allow display from a BytesIO object.

deshipu, 2021-06-01:
There is already the imageload library that lets you load images and display them from memory.

wifijt, 2021-06-01:
I looked at imageload https://circuitpython.readthedocs.io/projects/imageload/en/latest/api.html

That seems to only load from a file - I believe I tested this and found it was looking for a file object and didn't take the bytesIO object.

FoamyGuy, 2021-06-01:
+1, I would love to see a solution that allows displaying images that came from the internet in displayio without needing to have the CIRCUITPY mounted as writable.

It's not relevant to all cases, but if your device has an SDCard connected you are able to download images to the SDCard without needing to have the CIRCUITPY filesystem be writable. If your using a device with an SDCard slot, like a PyPortal or PyGamer, or if you can attach an external breakout for it that might be another option.

tannewt, 2021-06-01:
I think the best option is to fix ImageLoad to take any file-like io object.

dhalbert, 2022-02-18:
Closing this in favor of https://github.com/adafruit/Adafruit_CircuitPython_ImageLoad/issues/51.

closed 2022-02-18 by dhalbert