adafruit/circuitpython #4837: Extend deep sleep capability to SAMD port

maholli, 2021-06-01:
This issue is intended to scope and track the remaining work needed to extend [existing deep sleep API](https://learn.adafruit.com/deep-sleep-with-circuitpython/overview) (`alarm`) to the SAMD port.

## Current state

From discord:

dhalbert:
> main takeaways are:
> - figure out where to store a small amount of state while deep-sleeping so you can tell why you woke up (reconstruct Alarm object)
> - light sleep may do nothing more than the equivalent of time.sleep(). light sleep needs to leave all the devices on so the program can continue from where it left off
> - find out which pins you can use for deep-sleep wakeup and how they work (active low, active high, touch, etc.)
> - for the state storage, there are often some registers in the RTC or similar that can keep state during a deep sleep

tannewt:
> samd51 has backup ram as well
> the API for sleep is all set so that makes it easier
> debugging can be trickier though so I don't think adding support is particularly easy
> SAMD21 doesn't really have sleep modes so WFI is as good as you'll get IIRC

I can't guarantee I'll have the bandwidth to attempt this, but I'll start poking around. Hopefully others will want to contribute as well.

tannewt, 2021-06-02:
Here is an example of adding support to the RP2040 port: https://github.com/adafruit/circuitpython/pull/4816

It should be similar structurally for SAMD.

hierophect, 2021-06-03:
The common challenges, in my experience:

- Debugging, as Scott mentioned. Using a Saleae logic analyzer is useful, since you can manually yank some designated test pins up and down to help measure timing and detect problems.
- Making sure alarms correctly re-enable after triggering in light sleep, or don't trigger repeatedly/constantly.
- Determining the differences between viable triggers for deep sleep vs light sleep (often RTC is basically the same, but PinAlarm uses totally different calls)
- Converting chip monotonic time into RTC time for TimeAlarm. This might require some synchronization and moving between struct types. See STM32 and RP2040 for examples.
- Making sure PinAlarms are not inadvertently disabled when the program exits for deep sleep, due to `port_reset` resetting the GPIOs at the end of the VM. Typically either pin info must be stored and re-initialized, or the pins must be added to never_reset and manually reset in `alarm_reset`, which only occurs at the start of the *next* run
- Figuring out what woke the chip - this is really easy if the chip has some kind of `get_wake_reason()` but other chips need to store data in non-volatile "scratch" registers within the RTC or use other manual detection methods. It's possible that on some chips it actually won't be possible to tell.
- Making sure all 4 different sleep "types" are covered - both fake and true light sleep, and fake and real deep sleep ("fake" sleep means USB is attached and Circuitpython doesn't actually go to sleep at all). Deep sleep may need two different kinds of interrupts enabled simultaneously to prepare for either sleep variety.

[image]

PranavMKrishnan, 2021-07-01:
Hi, I'll be trying to implement the deep sleep functionality. I'm a newbie when it comes to the SAMD51 and CircuitPython in general. I have begun going through the code used for the deep sleep implementation on STM32 and would love to hear if there's a better way I should approach this project (maybe some basics I should know beforehand).

maholli, 2021-07-16:
I spent a evening attempting this and couldn't make enough progress and I wont have another opportunity for months. Unfortunately, @PranavMKrishnan gave it his best as well but wont be able to implement deep sleep either. :( 

@hierophect if I were to pool some money together into a "bounty," do you have the bandwidth to attempt this?

dxstp, 2021-07-21:
@maholli @PranavMKrishnan I have some low-level hardware experience with the D51, but struggle where to insert the needed hardware instructions in this project. Please contact me directly for details, maybe we can do this together.

maholli, 2021-07-21:
@dxstp I couldn't find your contact info anywhere on your github. If you're on [adafruit's discord](https://discord.gg/adafruit), that might be the easiest place to discuss. You can reach me at the same username: `maholli`, there as well. Otherwise, please send me an email. Address can be found here: https://maholli.com/

maholli, 2021-09-20:
OK finally got around to working on this.

I've made progress and can put a SAMD into deep sleep from circuitpython. However, I'm stuck getting **fake** deep sleep playing nice. 

Since https://github.com/adafruit/circuitpython/pull/5048 is closed, I've attached the relevant files so others can potentially help from here: [common-hal.zip](https://github.com/adafruit/circuitpython/files/7193260/common-hal.zip)

- cloning https://github.com/adafruit/circuitpython/pull/5048's commit and replacing the files from `/atmel-samd/common-hal/alarm/` with the (specific) attached zipped files will let you compile CP for the `feather_m4_express` (just add `CIRCUITPY_ALARM = 1` to the feather_m4's `mpconfigboard.mk` board def. 
  - only `TimeAlarm` is implemented. You'll find you can put the board into deep sleep so long as USB isn't connected. 
  - when USB is connected, it'll just loop through my `common_hal_alarm_pretending_deep_sleep()` print messages until halted. 
  - for some reason, I wasn't successful when attempting to put the `RTC` register twiddling where it belongs in `alarm_time_timealarm_set_alarms()`. The result are some inefficient hacks to TimeAlarm.c but it gets the job done for the time being.
  
@hierophect do you have a minute to provide some insight?

tannewt, 2021-09-20:
Please provide a comparison link to the changes (your branch vs main) rather than requiring local modifications.

maholli, 2021-09-20:
> 
> 
> Please provide a comparison link to the changes (your branch vs main) rather than requiring local modifications.

maholli, 2021-09-20:
> 
> 
> Please provide a comparison link to the changes (your branch vs main) rather than requiring local modifications.

Updated the comment to reference a proper commit change log. Thanks for the reminder.

tannewt, 2021-09-21:
To some degree it should loop through pretending to deep sleep because main will loop until an alarm was actually detected. It'll only sleep between interrupts: https://github.com/adafruit/circuitpython/blob/main/main.c#L533

maholli, 2021-09-24:
ok well true deep sleep and fake deep sleep is working in this commit: https://github.com/maholli/circuitpython/commit/f030a522353b60e535d756d63fcc2d93d5c8d91e 

fake sleep is a little janky. I haven't been able to get a CMP flag to trigger while the vm is running, so for now it just checks `port_get_raw_ticks` while looping in fake sleep.

hierophect, 2021-09-26:
Ok I am WAY LATE to the party here, sorry @maholli, but here is a summary of the existing ways the different ports handle fake sleep. Hopefully some of this will be helpful - I'm still not totally grokking the way the ASF4 is handling these interrupts.

- **STM32**: These chips are pretty straightforward, because the RTC alarm feature, which compares the current RTC time to a calendar date by default, will generate interrupts successfully regardless of whether the MCU is in "run" or "sleep" mode. When the TimeAlarm is set with `alarm_time_timealarm_set_alarms`, an alarm interrupt is set in the future, and a callback is created that will be called with `HAL_RTCEx_AlarmBEventCallback`. If the deep sleep is "fake", this callback will be executed (setting `woke_up`, returned with `alarm_time_timealarm_woke_this_cycle`), and `alarm_get_wakeup_cause` will use it to determine the wakeup was a timealarm. If it is a true deep sleep, there won't be a callback and the chip will wake as if just starting up, using flags it reads out of the backup registers and `RTC->ISR` to determine if it was a GPIO or RTC that woke the chip. One other note, however, is that the time set for deep sleep could be shorter than the time it takes to detect if USB is connected, which is used to determine whether sleep is real or fake - to avoid this, `alarm_time_timealarm_prepare_for_deep_sleep` (only called before true deep sleep) will automatically override the existing alarm interrupt before true deep sleeping (it just sets it again, basically). Note that PinAlarms are a smidge more complicated because they involve `common_hal_alarm_pretending_deep_sleep`, but TimeAlarms don't interact with that function.
 - **ESP32S2**: on this chip, two completely different methods are used for fake vs true deep sleep. When `alarm_time_timealarm_set_alarms` is called, it calculates ticks and inputs them into both `esp_sleep_enable_timer_wakeup` (used for true deep sleep) and `esp_timer_start_once` (used for fake deep sleep). If the chip true deep sleeps, it will call `esp_deep_sleep_start` from `common_hal_alarm_enter_deep_sleep`, which will use the time from `esp_sleep_enable_timer_wakeup` and wipe out/ignore the timer value when the RAM is dropped. If it fake deep sleeps, since `esp_deep_sleep_start` is never called, the `esp_sleep_enable_timer_wakeup` is ignored and the chip will wake up once `esp_timer_start_once` goes off and `woke_up` is set by `timer_callback`, detected by `common_hal_alarm_woken_from_sleep` in main. ESP32S2 does not use `common_hal_alarm_pretending_deep_sleep` at all.
 - **NRF**: these chips don't actually use interrupts at all, but instead use a polling method in both the deep and light sleep case. `alarm_time_timealarm_set_alarms` just sets a global time value that will be used for either light or deep sleep, and the "true" deep sleep in `common_hal_alarm_enter_deep_sleep` just turns the clock speed down a bit before calling `system_on_idle_until_alarm`. (I didn't write this one so I'm not totally sure about it).
  - **RP2040**: this one is also weird, because the RP2040 is incapable of dropping RAM in deep sleep. It's also unable to use TimeAlarm in the lowest power mode without an external pin providing a clock, so if a TimeAlarm exists, it actually just enters more of a very deep light sleep. But in any case, it uses the same interrupt for everything, set by `rtc_set_alarm`, and then just resets the MCU after sleeping if it was a "true" deep sleep.
  
 TLDR:

-  STM32: uses the same interrupt for both real and fake, with protection against really short interrupts.
-  ESP32S2: uses different timers for real and fake, deep sleep overriding the light sleep one.
-  NRF: No interrupts, just a global time that both methods poll.
-  RP2040: Both methods are actually just light sleep, and use the same interrupt, but one resets afterwards.

hierophect, 2021-09-26:
@maholli your current method seems fine, tbh, close to the STM32 implementation. Fake deep sleep needs sleep performance the least out of any mode, because the assumption is that you'll only ever use it when debugging and connected to a PC. But if we want to do light sleep (where RAM isn't dropped), it would be good to figure out how to get that CMP to work.

Are you currently able to detect, after waking from true deep sleep, whether it woke up from sleep or just started for the first time? Is there a register that you can check that will tell you it woke from the TimeAlarm? Once Pin Alarm gets added, we'd also want to be able to tell the two kinds of wakeup apart. In fake deep sleep, you should also be able to tell them apart based on your manual delay loop (or more ideally a callback) setting `woke_up` per module. Left you some comments directly as well.

maholli, 2021-09-28:
Thanks for your insight @hierophect! Yes, I'm able to detect whether wake-up is from POR, BACKUP, etc...

I've continued to wrestle with getting CMP to fire an interrupt, but have been unsuccessful. I'm convinced there's some micropython/circuitpython interrupt handler I can't find that's quietly clearing the flag.

@tannewt are there any interrupt handlers or MP/CP things I could be missing that'd impact an RTC interrupt during fake sleep like this? Below is a print-out of relevant RTC registers the [moment my manual check](https://github.com/maholli/circuitpython/blob/f030a522353b60e535d756d63fcc2d93d5c8d91e/ports/atmel-samd/common-hal/alarm/__init__.c#L229) catches `COUNT` > `COMP1`. As you can see, `COUNT` is greater than `COMP1`, but the `INTFLAG` bit isn't set.

```c++
COUNT 8456
CTRLA 33282
    count sync enabled; prescalar=DIV2; RTC enabled
CTRLB 0
EVCTRL 0
INTENCLR 33280
    overflow; COMP1
INTENSET 33280
    overflow; COMP1
INTFLAG 255
    all PER flags but no COMP or overflow flags
SYNCBUSY 0
COMP0 69743
COMP1 8455
```

tannewt, 2021-09-28:
How do you know it isn't getting into RTC_Handler? https://github.com/adafruit/circuitpython/blob/main/ports/atmel-samd/supervisor/port.c#L477

maholli, 2021-09-28:
> 
> 
> How do you know it isn't getting into RTC_Handler? https://github.com/adafruit/circuitpython/blob/main/ports/atmel-samd/supervisor/port.c#L477

@tannewt I've checked by added `printf` statements to `RTC_Handler` to check for COMP0 and COMP1. Only time I get something is when COMP0 fires for a sleep (for example).

tannewt, 2021-09-28:
Two other things to check would be GP2EN in CTRLB and also the errata. It's possible it's broken.

maholli, 2021-09-29:
TimeAlarm and PinAlarm are now both working on https://github.com/maholli/circuitpython/commit/d536be7228eac78b95e0b838110997434bcdad7a. 

But there are caveats!

- PinAlarm uses only the `TAMPER` pins for deep sleep. PinAlarm is also working with light sleep, but that's achieving the same power consumption as `time.sleep()` so we may want to remove it.
- There are only a couple `TAMPER` pins:
   [image]
   - Now there ARE errata for the RTC tamper functions. Mainly this:
     [image]
     <br>
     I haven't experienced this, but that's not to say it wont pop up. Here are the workarounds:
     [image]
- fake sleep with `TimeAlarm` still has to manually check the CMP register. This may be fine?

tannewt, 2021-09-30:
I think it's a good start. Ready for another review?

maholli, 2021-09-30:
> 
> 
> I think it's a good start. Ready for another review?

Not quite. I'll ping you when I get things cleaned up. Do you want that to be a PR?

dhalbert, 2022-10-18:
Most fixed by #5425. An issue for any remaining functionality can be opened.

closed 2022-10-18 by dhalbert