adafruit/circuitpython #4838: Raspberry Pi Pico - Soft reset doesn't clear out pin usage state

dannystaple, 2021-06-01:
**Firmware**

Adafruit CircuitPython 6.2.0 on 2021-04-05; Raspberry Pi Pico with rp2040
```python
Adafruit CircuitPython 6.2.0-beta.2 on 2021-03-01; Raspberry Pi Pico with rp2040
```

**Code/REPL**

```python
import board
import displayio
import busio
from adafruit_st7789 import ST7789

spi = busio.SPI(board.GP18, MOSI=board.GP19)
while not spi.try_lock():
    pass
spi.configure(baudrate=24000000) # Configure SPI for 24MHz
spi.unlock()
tft_cs = board.GP17
tft_dc = board.GP16

displayio.release_displays()
display_bus = displayio.FourWire(spi, chip_select=tft_cs, command=tft_dc)

display = ST7789(display_bus, width=240, height=240, rowstart=80, rotation=180)

print("Hello on the display!")
```

**Behavior**

It runs the first time, but if I make changes and save (causing soft reset) or Ctrl-D for a soft reset, I get the following:

```python
code.py output:
Traceback (most recent call last):
  File "code.py", line 5, in <module>
ValueError: GP18 in use
```

**Description**

Only seen so far with this SPI setup - but this is my first CircuitPython/Pico SPI device usage.

- might be related to #352...

tannewt, 2021-06-01:
Displays hold onto their pins so that they live through the reset. You'll want a `displayio.release_displays()` before you reconstruct it.

tannewt, 2021-06-01:
More specifically, you'll want it before you reconstruct the SPI.

closed 2021-06-01 by tannewt

dannystaple, 2021-06-02:
Thanks. It's a shame I can't use atexit for this. I'm thinking mini-game code that potentially has a big while loop, and I'm assuming the soft exit doesn't come through as an exception that I could handle with a finally clause. Are there any good patterns for that?

tannewt, 2021-06-03:
> Thanks. It's a shame I can't use atexit for this. I'm thinking mini-game code that potentially has a big while loop, and I'm assuming the soft exit doesn't come through as an exception that I could handle with a finally clause. Are there any good patterns for that?

The MicroPython merge brought in `atexit` functionality but they lumped it into `sys` so I disabled it. It shouldn't be too hard to move and enable.

You may be interested in #3454 which was done with game code in mind too. It has mechanics for what to do after an exception.