adafruit/circuitpython #4839: rp2040: every other reading incorrect with simpletest

jerryneedell, 2021-06-01:
When I run the hcsr04_simpletest on a feather_rp2040, every other reading is very short. The ~40cm reading is correct.
This works normally on a feather_m4_express

I am using the RCWL-1601 sensor

```

Adafruit CircuitPython 7.0.0-alpha.2-705-gc80e25307 on 2021-05-31; Adafruit Feather RP2040 with rp2040
>>> 
>>> import hcsr04_simpletest
(44.047,)
(1.411,)
(43.911,)
(1.326,)
(43.911,)
(1.071,)
(43.197,)
(1.768,)
(42.415,)
(1.7,)
(42.143,)
(1.479,)
(39.355,)
(1.445,)
(41.837,)
```

tannewt, 2021-06-02:
I think @DavePutz would know about the current PulseIn state.

DavePutz, 2021-06-03:
Instrumentation shows that the issue is that occasionally the PIO input from the Pico is returning a zero bit in a result word when it should be a one.  As a result, the code detects that as an edge change and returns the short value. The logic analyzer confirms that the pulse coming in from the HCSR04 does not have any drops in the signal. Changing pins and voltages does not have any effect on the issue. Will pursue what might be done to correct this.

jerryneedell, 2021-06-04:
fixed by #4859

closed 2021-06-04 by jerryneedell