adafruit/circuitpython #4842: Nano Connect RP2040 NeoPixel Issues

dedSyn4ps3, 2021-06-02:
**Firmware**

```python
Adafruit CircuitPython 6.3.0-rc.0 on 2021-05-25; Arduino Nano RP2040 Connect with rp2040
```

**Code/REPL**

```python
"""CircuitPython Essentials NeoPixel example"""
import time
import board
import neopixel

pixel_pin = board.D9
num_pixels = 8

pixels = neopixel.NeoPixel(pixel_pin, num_pixels, brightness=0.3, auto_write=False)

def wheel(pos):
    # Input a value 0 to 255 to get a color value.
    # The colours are a transition r - g - b - back to r.
    if pos < 0 or pos > 255:
        return (0, 0, 0)
    if pos < 85:
        return (255 - pos * 3, pos * 3, 0)
    if pos < 170:
        pos -= 85
        return (0, 255 - pos * 3, pos * 3)
    pos -= 170
    return (pos * 3, 0, 255 - pos * 3)

def color_chase(color, wait):
    for i in range(num_pixels):
        pixels[i] = color
        time.sleep(wait)
        pixels.show()
    time.sleep(0.5)

def rainbow_cycle(wait):
    for j in range(255):
        for i in range(num_pixels):
            rc_index = (i * 256 // num_pixels) + j
            pixels[i] = wheel(rc_index & 255)
        pixels.show()
        time.sleep(wait)

RED = (255, 0, 0)
YELLOW = (255, 150, 0)
GREEN = (0, 255, 0)
CYAN = (0, 255, 255)
BLUE = (0, 0, 255)
PURPLE = (180, 0, 255)

while True:
    pixels.fill(RED)
    pixels.show()
    # Increase or decrease to change the speed of the solid color change.
    time.sleep(1)
    pixels.fill(GREEN)
    pixels.show()
    time.sleep(1)
    pixels.fill(BLUE)
    pixels.show()
    time.sleep(1)

    color_chase(RED, 0.1)  # Increase the number to slow down the color chase
    color_chase(YELLOW, 0.1)
    color_chase(GREEN, 0.1)
    color_chase(CYAN, 0.1)
    color_chase(BLUE, 0.1)
    color_chase(PURPLE, 0.1)

    rainbow_cycle(0)  # Increase the number to slow down the rainbow

```

**Behavior**

Code seems to run just fine with no traceback errors. NeoPixels fail to light up even when using multiple other pins.

**Description**

- Error while attempting to use neopixel strand with the nano connect

**Additional Info**

Have had zero issues using normal pin output for regular LEDs using DigitalInOut and board definition for multiple different pins on the board. Single LED works just fine, but as soon as you instantiate a NeoPixel object on a given pin, the code will run...but the NeoPixel strand does nothing :'(

...and just to throw it in for kicks, yes the neopixel strand works just fine. Used it on three other boards running CP, all good. Not sure if it has anything to do with the __init__ used by the library for creating neopixel objects or not...but they clearly are having a issue running with this board right now.

It does not seem limited to just circuitpython either. I have tried running multiple NeoPixel programs in C/C++ using the Arduino IDE and NP examples, as well as BLE NeoPixel control. The programs compile, load, and run....just no NeoPixel response...
[image]

ladyada, 2021-06-02:
if it doesnt work in arduino or circuitpython its probably not a firmware thing. try powering the NeoPixel strip from 3.3V or adding a level shifter. you should probably invest in a scope or logic analyser as well

dedSyn4ps3, 2021-06-02:
### **@ladyada yes a good point! I tried using the 3.3V pin for power using a different version of the test code above, which utilized a different pin and had no luck.**

After looking over the detailed pinout I saw that a number of the digital (and analog) pins are routed through the Nina Wifi/BLE module which I'm sure I've seen you mention elsewhere.

**I have tested using the 3.3V and the same pin above, D9, just now and have gotten the NeoPixel test code to run properly with the proper animations...fantastic!**

closed 2021-06-03 by ladyada