adafruit/circuitpython #4854: MicroMod SAMD51 running in safe mode with heap allocation error

stonehippo, 2021-06-03:
**Firmware**

```python
Adafruit CircuitPython 7.0.0-alpha.2-755-g9cde8a226 on 2021-06-03; SparkFun MicroMod SAMD51 with samd51j20
```

**Behavior**

When connecting the MicroMod SAMD51, the CIRCUITPY drive does not mount, but I can still access the REPL via the serial port. The system is running in safe mode.

```python
Auto-reload is off.
Running in safe mode! Not running saved code.

You are in safe mode because:
CircuitPython core code crashed hard. Whoops!
Attempted heap allocation when VM not running.
Please file an issue with the contents of your CIRCUITPY drive
```

Running simple code (such as configuring the UART as follows) seems to work fine from the REPL.

```python
import board
import busio

uart = busio.UART(board.TX, board.RX, baudrate=9600)
```

I recently added this board support, so I'm sure I missed something in the config, but I'm not sure what. I've tested the same 7.0.0-alpha on a Sparkfun SAMD51 Thing Plus without any issues.

tannewt, 2021-06-03:
Triple check your SPI flash pinout. If it was trying to construct a static busio.SPI on the wrong pins it'll try to raise an exception which requires an allocation.

stonehippo, 2021-06-03:
Yeah, checked it against the schematic a couple of times, but will do it again.

lesamouraipourpre, 2021-06-03:
The pins that talk to the Flash aren't defined in pins.c
PA08, PA09, PA10, PA11, PB22, PB23

I don't know what names they need to be defined with in pins.c.

Ref: https://cdn.sparkfun.com/assets/1/d/3/e/b/SparkFun_MicroMod_SAMD51_Processor_Board_Schematic.pdf
[image]

lesamouraipourpre, 2021-06-03:
Ignore me.

They are defined in mpconfigboard.h and the naming matches.
Are the two extra pins !WP(PB22) and !HOLD(PB23) required for operation?

stonehippo, 2021-06-03:
@lesamouraipourpre I don't know. All the board dead I've looked at only needed the basic SPI pins (for example, I only defined those four with the MicroMod RP2040, which uses the same external flash).

I've been wondering if there's a way to see the crash. The system recovers enough to give me safe mode and a REPL, so the flash chip seems likely, but it's hard to operate on guesses. ;-)

tannewt, 2021-06-04:
> I've been wondering if there's a way to see the crash. The system recovers enough to give me safe mode and a REPL, so the flash chip seems likely, but it's hard to operate on guesses. ;-)

The easiest way is to connect a debugger and set a breakpoint on `reset_into_safe_mode` that'll stop you before the reset.

stonehippo, 2021-06-04:
Thanks @tannewt. I'll give that a try this weekend.

stonehippo, 2021-06-05:
@tannewt @dhalbert what does the `SPI_FLASH_FILESYSTEM = 1` flag in do in `mpconfigboard.mk`? It was in the Sparkfun Thing Plus board definition, and I left it in when I created the MicroMod def, but I just realized it's not always present. I didn't have the MicroMod RP2040 def, for example. I'm wondering if the issue here is that this needs to be swapped for `QSPI_FLASH_FILESYSTEM`.

stonehippo, 2021-06-05:
OK, yeah, I think it's the `SPI_FLASH_FILESYSTEM` flag. I the nvm.toml repo and the Sparkfun SAMD51 Thing Plus uses an AT25SF041A, which is SPI, and the MicroMod is using a W25Q128JVPM, which is QSPI. I'm going to try making that change and building.

stonehippo, 2021-06-05:
I tried using `QSPI_FLASH_FILESYSTEM`, removed the SPI flash pins definitions, and added PB22 and PB23 to the ports so they wouldn't get reset. Result was worse (crashed hard enough to keep me from the REPL) but I think I know why.

The atmel-samd port hardcodes the pins in `supervisor/qspi_flash.c`:

```c
// The QSPI is only connected to one set of pins in the SAMD51 so we can hard code it.
uint32_t pins = {PIN_PA08, PIN_PA09, PIN_PA10, PIN_PA11, PIN_PB10, PIN_PB11}
```

The MicroMod does _not_ have `PIN_PB10` or `PIN_PB11` connected to the flash module; it's using `PIN_PB22` and `PIN_PB23` instead, which is probably why stuff blew up when I switched to QSPI. I'm gong to check the SAMD51 datasheet to see if it's possible to use those pins for the QSPI interface. If so, I'll probably path `qspi_flash.c` to allow for setting the pins and give this another try.

stonehippo, 2021-06-05:
Yep, that did it. Switching to QSPI and changing the pins let me boot the board without dropping into safe mode. I'm not seeing the CIRCUITPY volume mount when I connect the device or reset, though. I guess I'll tackle that after I can something in to allow pin selection for QSPI on the atmel-samd port.

lesamouraipourpre, 2021-06-07:
Hi.

I've been reading the Datasheet and the SparkFun Github and the Flash on the SAMD51-Micromod is not QSPI:
https://github.com/sparkfun/MicroMod_Processor_Board-SAMD51/commit/85ee4311319f440b5569dbe4685bf731b0e9c310
```
* QSPI was removed in place of a secondary SPI port
* ...
```

Looking at other boards with SPI Flash the WP and HOLD pins are tied to VCC.
Examples:
https://cdn-learn.adafruit.com/assets/assets/000/040/553/large1024/arduino_schem.png?1490994398
https://cdn.sparkfun.com/assets/8/2/1/d/f/SAMD51_Thing_Plus_v10.pdf

I suspect that you will need to pull PB22 & PB23 high at boot. I've attempted to make something which _might_ work - I've not done any boot initialisation development so this is wild-ass guessing. There may be a better way to do this. I've attached a diff file based off the current `main` branch. It builds but I have no way of testing it or even being sure if what I have done is correct.
[samd51-micromod.diff.txt](https://github.com/adafruit/circuitpython/files/6611732/samd51-micromod.diff.txt)

stonehippo, 2021-06-07:
@lesamouraipourpre I'll give this a try. It seems that the flash chip can run in SPI or QSPI, and is suspect that PB22/23 work fine for that purpose, as they are SERCOM capable. But I'd rather keep it simple and mimic what they did with the Arduino variant, as you suggest.

dhalbert, 2021-11-16:
https://github.com/adafruit/circuitpython/pull/5565 notes that there is a design bug on v1.2 of this board: The MOSI and MISO pins to the external flash were switched. Also see https://github.com/sparkfun/MicroMod_Processor_Board-SAMD51/issues/2, which notes they hope to have fixed ones in stock. I'll open a new issue about fixing the pin mapping; let's close this one.

closed 2021-11-16 by dhalbert