adafruit/circuitpython #4866: kaluga: pinout matches board rev 1.2, current is rev 1.3

jepler, 2021-06-07:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/hw-reference/esp32s2/user-guide-esp32-s2-kaluga-1-kit.html#hardware-revision-details

> The following pins re-assigned to fix the download issue
> * Camera D2： GPIO36
> * Camera D3： GPIO37
> * AU_I2S1_SDI： GPIO34
> * AU_WAKE_INT： GPIO46

At least from Adafruit, the last 2 modules I received were revision 1.3.  Revision 1.3 was noted in esp-idf's git in August 2020. (Revision 1.2 was the first public revision and I had 2 modules of that vintage, which may have been from adafruit or from digikey)

I don't know which of the following should be done:
 * Add a new board definition for 1.3 (espressif_kaluga_1.3 or …1_3), leaving the existing definition
 * Rename the current board definition 1.2 (espressif_kaluga_1.2 or …1_2), adding a new board definition for 1.3
 * Just modify the single board definition, so that only 1.3 is supported.
The board can be identified by the silk on the bottom.  The box it ships in does not reflect the revision number that I could see.

jepler, 2021-06-07:
@tannewt an opinion would be appreciated.

jerryneedell, 2021-06-07:
Just my opinion. It looks like the v1.3 should be more common. (that is what I received but I don't recall when or from where I got it...)  I'd like to see 1.3 supported either as the default or as a new version.
I have not tried CP on it yet so you can tell it is not a high priority for me.

tannewt, 2021-06-07:
My preference would be to add a new board with the version number (1.3) in the name.

PR #4870 (2021-06-09): Add kaluga version 1.3
.. this board is nearly the same as the "1.2" version originally released, but makes a few changes to reduce pin conflicts between the 13-pin camera header and the bootstrapping pins.  "1.3" was introduced in summer 2020-- check the bottom of your board silk to find out whether you have a 1.2 or a 1.3, because it's not clear whether there is remaining 1.2 stock.

Another vexing fact about the 1.3 is that its LCD can have one of two chipsets, and the chipset used is not marked anywhere! I literally got one of each in my last two orders.

Closes: #4866
closed 2021-06-09 by jepler