adafruit/circuitpython #4872: BLE write from nRF52840 to Pyportal or Matrix Portal hangs CP

wifijt, 2021-06-09:
6.2 on nRF52840 and PyPortal and Matrix Portal

Adafruit CircuitPython 6.2.0 on 2021-04-05; Adafruit PyPortal with samd51j20

```python
Adafruit CircuitPython 6.2.0-beta.2 on 2021-03-01; Raspberry Pi Pico with rp2040
```

**Code/REPL**

Using the echo client example - modified for the ESP32

from adafruit_ble import BLERadio
from adafruit_ble.advertising.standard import ProvideServicesAdvertisement
from adafruit_ble.services.nordic import UARTService
from adafruit_airlift.esp32 import ESP32
esp32 = ESP32()
adapter = esp32.start_bluetooth()

ble = BLERadio(adapter)

**Behavior**

The echo write hangs the nRF52840

also recreated on the REPL

uart_connection[UARTService].write('foo')
```python
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TimeoutError: Clock stretch too long
```

**Description**

I am trying to send UART from the nRF52840 to a matrix portal - however whenever a packet gets sent the nRF52840 just hangs - I cannot ctrl-C - if I kill the Matrix portal or pyportal the nRF52840 will go back to scanning.

This is easy to replicate using the echo test between the nRF52840 and pyportal

**Additional Info**

tannewt, 2021-06-09:
Please attach your full code. "Clock stretch" is usually a term used with I2C, not UART.

wifijt, 2021-06-09:
Sorry that was left over from the bug report template. 

This is the code on the MatrixPortal

`

"""
Used with ble_uart_echo_client.py. Receives characters from the UARTService and transmits them back.
"""

from adafruit_ble import BLERadio
from adafruit_ble.advertising.standard import ProvideServicesAdvertisement
from adafruit_ble.services.nordic import UARTService
from adafruit_airlift.esp32 import ESP32

esp32 = ESP32()
adapter = esp32.start_bluetooth()

ble = BLERadio(adapter)
uart = UARTService()
advertisement = ProvideServicesAdvertisement(uart)

while True:
    ble.start_advertising(advertisement)
    while not ble.connected:
        pass
    while ble.connected:
        print(uart.read(1))
        # Returns b'' if nothing was read.
        one_byte = uart.read(1)
        if one_byte:
            print(one_byte)
            uart.write(one_byte)
`

This is the code on the nRF52840

'"""
Used with ble_uart_echo_test.py. Transmits "echo" to the UARTService and receives it back.
"""

import time

from adafruit_ble import BLERadio
from adafruit_ble.advertising.standard import ProvideServicesAdvertisement
from adafruit_ble.services.nordic import UARTService

ble = BLERadio()
while True:
    while ble.connected and any(
        UARTService in connection for connection in ble.connections
    ):
        for connection in ble.connections:
            if UARTService not in connection:
                continue
            print("echo")
            uart = connection[UARTService]
            uart.write(b"echo")
            # Returns b'' if nothing was read.
            one_byte = uart.read(4)
            if one_byte:
                print(one_byte)
            print()
        time.sleep(1)

    print("disconnected, scanning")
    for advertisement in ble.start_scan(ProvideServicesAdvertisement, timeout=1):
        if UARTService not in advertisement.services:
            continue
        ble.connect(advertisement)
        print("connected")
        break
    ble.stop_scan()

'

silviaCC, 2021-07-16:
I have replicated this issue, with the same code, on a Circuit Playground Bluefruit writing to a Pyportal Titano 
(CircuitPython 6.3.0).
The code hangs on the CPB when trying to verify if `UARTService in connection`, and if I turn off the Pyportal, the code on CPB crashes with this traceback:   
`File "adafruit_ble/__init__.py", line 75, in __contains__`
`File "adafruit_ble/__init__.py", line 52, in _discover_remote`

Same code works correctly when connecting 2 Circuit Playground Bluefruit boards. 

The code on the Pyportal also works correctly when using it to connect with the Bluefruit Connect App.