adafruit/circuitpython #4874: Wrong filesystem size on Arduino RP2040 Connect

tannewt, 2021-06-09:
**Firmware**

All with Arduino RP2040 Connect

**Description**

Filesystem size is 1MB instead of 15MB.

My guess is that the flash chip isn't correctly reporting its size. So, we hit the default case of 1 MB. Related code is here: https://github.com/adafruit/circuitpython/blob/main/ports/raspberrypi/supervisor/internal_flash.c#L67-L79

The fix would be to use the flash size from nvm.toml if the value is out of range.

fivdi, 2021-06-23:
Hi, I ended up here in this issue after noticing the file system size for the Arduino Nano RP2040 Connect is 1 MB rather than the expected 15 MB. Of course, this also means it's not possible to copy more that 1 MB to the file system so the remaining 14 MB go unused 😄 

> My guess is that the flash chip isn't correctly reporting its size.

I wrote a small pico-sdk application to see what the call to `flash_do_cmd` [here](https://github.com/adafruit/circuitpython/blob/6.3.0/ports/raspberrypi/supervisor/internal_flash.c#L70) returns in array `data`. Here is what I saw in the array:
```
0xff 0x1f 0x89 0x01
```
This is what the datasheet says it should return. See section 7 on page 14 of the data sheet [here](https://www.mouser.de/datasheet/2/590/adesto_technologies_adet-s-a0007947218-1-1764155.pdf) for further details.

The datasheet refers to the "device identification" instruction (0x9f) as being a legacy instruction which somehow implies that it's no longer the best of ideas to use it, at least that is how I interpret the word legacy in this case.

Anyway, byte 3 in the array `data` is 0x01 which is the important byte for the calculation performed after calling `flash_do_cmd`. Unfortunately, 0x01 for the AT25SF128A flash memory used by Arduino Nano RP2040 Connect doesn't appear to mean the same thing as 0x01 for other flash memories on other RP2040 boards.

> The fix would be to use the flash size from nvm.toml if the value is out of range.

The pico-sdk has a define called PICO_FLASH_SIZE_BYTES, maybe this could be used? Then again, maybe not, I'm not sure how tightly coupled CircuitPython and pico-sdk are.

tannewt, 2021-06-24:
We generally don't use the pico-sdk for board defines because we supported more out of the box. Both GigaDevice and Winbond flashes will report 0x21+ for capacity.

For this, I think we'll want a small Python script similar to https://github.com/adafruit/circuitpython/blob/main/tools/gen_nvm_devices.py that generates a header file that either allows for the current detection or a fixed flash size. Then,the detection code could be overridden.

PR #5035 (2021-07-21): Fix Arduino RP2040 flash size
For RP2040 boards, we now change the default flash size based on
the configured flash. We will still try to read the size from the
flash first.

Fixes #4874
commit 109bd288 (2021-07-21): Fix Arduino RP2040 flash size

For RP2040 boards, we now change the default flash size based on
the configured flash. We will still try to read the size from the
flash first.

Fixes #4874
closed 2021-07-21 by dhalbert